zalo/CascadeStudio #4: RuntimeError: function signature mismatch

ancorasir, 2020-07-30:
I use the following code to generate a list of TopoDS_Shape brepList

```
function ForEachShell(shape, callback) {
  let shell_index = 0;
  let anExplorer = new oc.TopExp_Explorer(shape, oc.TopAbs_SHELL);
  for (anExplorer.Init(shape, oc.TopAbs_SHELL); anExplorer.More(); anExplorer.Next()) {
    callback(shell_index++, anExplorer.Current());
  }
}
```
Then modify one of the shape in brepList like
`brepList[someIndex] = Rotate(axis = [0, 1, 0], degrees = 1, somebrep);
`
and the following code to make an compound
```
let scene = new oc.TopoDS_Compound();
let sceneBuilder = new oc.BRep_Builder();
sceneBuilder.MakeCompound(scene);
brepList.forEach((curShape) => {
                  console.log(curShape)
                  sceneBuilder.Add(scene, curShape);
});
this.updateShape(scene, 0.01 )
```

I got Uncaught RuntimeError: function signature mismatch at BRep_Builder.Add.BRep_Builder.Add (http://127.0.0.1:5500/node_modules/opencascade.js/dist/opencascade.wasm.js:11:914335)

Any suggestion? Am I misuse the functions? I print the curShape and it looks fine...
@zalo

zalo, 2020-07-31:
Instead of calling `this.updateShape()` yourself, consider simply adding the shapes you want to draw to the global `sceneShapes` array.   The backend will create a compound shape with a BRep_Builder using sceneShapes automatically.

So something like:
```ForEachShell(shape, (index, curShape) => sceneShapes.push(curShape));```

Also consider checking whether your `curShape.IsNull()`.    I haven't done any testing with `TopoDS_Shell`s yet, so you're the first to test whether they work!

ancorasir, 2020-07-31:
I tested the following code

```
let holeRadius = Slider("Radius", 30 , 20 , 40);

let sphere     = Sphere(50);
let cylinderZ  =                     Cylinder(holeRadius, 200, true);
let cylinderY  = Rotate([0,1,0], 90, Cylinder(holeRadius, 200, true));
let cylinderX  = Rotate([1,0,0], 90, Cylinder(holeRadius, 200, true));

Translate([0, 0, 50], Difference(sphere, [cylinderX, cylinderY, cylinderZ]));

// Don't forget to push imported or oc-defined shapes into sceneShapes to add them to the workspace!
console.log("Check brep=====================");
const brepList1 = [];
const brepList2 = [];

brepList1.push(sceneShapes[0]);

let anExplorer = new oc.TopExp_Explorer(sceneShapes[0], oc.TopAbs_SHELL);
for (anExplorer.Init(sceneShapes[0], oc.TopAbs_SHELL); anExplorer.More(); anExplorer.Next()) {
    myShell = oc.TopoDS.prototype.Shell(anExplorer.Current());
    console.log("Shell inside for loop is not ", myShell.IsNull())
    brepList2.push(myShell);
};

console.log("brep1 is not ",brepList1[0].IsNull());
console.log("brep2 is not ",brepList2[0].IsNull());
```

The outpu is 
`Check brep=====================
index.js:251 Shell inside for loop is not  false
index.js:251 brep1 is not  false
index.js:251 brep2 is not  true`

Seems I cannot get a list of shells by pushing the shells inside the for loop. @zalo

zalo, 2020-08-05:
Hrm, I now believe that Solids do not contain Shells, just as Faces do not contain the Wires that could have gone into making them.  The two generally iterable types appear to just be Faces and Edges.

Therefore, I believe the inability to iterate across Shells in the Sphere primitive is generally correct behavior in the context of OpenCascade, and I'll be closing this issue (unless you can find an example of someone attempting this in the C++ version and succeeding...)

ancorasir, 2020-08-07:
Actually I succeeded in getting the shell of the sphere using the following code

`let holeRadius = Slider("Radius", 30 , 20 , 40);

let sphere     = Sphere(50);

function getShell(shape){
    let anExplorer = new oc.TopExp_Explorer(shape, oc.TopAbs_SHELL);
    for (anExplorer.Init(shape, oc.TopAbs_SHELL); anExplorer.More(); anExplorer.Next()) {
        return oc.TopoDS.prototype.Shell(anExplorer.Current());
    };
};

shell = getShell(sphere);
shell1 = Translate([0,0,50], shell);
sceneShapes.push(shell1);
`

I can see two spheres in the scene. The strange part is I can get the shell using "return shell" but not ".push(shell)" out side the for loop...
[image]

zalo, 2020-08-07:
Interesting; I notice you’re also casting it to a shell object... I’ll need to investigate more carefully...

ancorasir, 2020-08-08:
This problem also occurs to face as well.

```
function getFace(shape){
    let anExplorer = new oc.TopExp_Explorer(shape, oc.TopAbs_FACE);
    var list = [];
    for (anExplorer.Init(shape, oc.TopAbs_FACE); anExplorer.More(); anExplorer.Next()) {
        ff = oc.TopoDS.prototype.Face(anExplorer.Current());
        list.push(ff);
    };
    return [ff, list];
};
```

The above code will return null face and list of null face. But if put the return inside the for loop, I will get face and list of face correctly. I guess the problem is about how js get access to the webassembly memory. Somehow, outside `for`/`if`, the pointer to the memory get ruined. I'm not familiar with this part...

zalo, 2020-08-09:
Unfortunately, neither am I; perhaps @donalffons knows more about how the memory/references for shapes are allocated and handled?  Perhaps TopExp just needs an IDL modification?  Or the references need to be grabbed via special Handle wrapper types?

My current workaround suggestion is to do any work on the faces/shells from within the for-loop; I believe it’s presently how the ForEachEdge is able to handle adding fillets.

Though, I think I’m noticing a sister to this bug in the STEP file reader, where the shape from the last file that is read in overwrites the shape from the first file (preventing more than one step from being loaded...)

donalffons, 2020-08-10:
This seems to be an issue with how 

Here is a slightly modified code example from you @ancorasir ([CascadeStudio Example](https://zalo.github.io/CascadeStudio/?code=lZJda8IwFIbv%2BysO3hgh1O7DizF2MVSYIF5YLzZFRqynthiTkqRM%2F%2F1ObLXixj5CKWnOOc%2F75qUSHWRa4lSs89LCE8QyX6NhreqgxeEuAg63%2FnUfdR6DQNKELTI0CH7RxPGD9XzVF5ODzBUx5r743erXDawR9goRB2dKvKK8ecpUO%2BGQLSJ%2Bw6Mlhwfq%2FYVyhXm9xBDkH5hgZoSysjYA9PT88CBPU7q3SpBVcXBYnNV447%2FZzpedU4BCDfeF1AYN2VL4AToJZ7qgw%2FdTgdkEFcaZKNAuvGDV8ryy7%2FHLcDz2qAYTjlTu%2FjCSaGUd7A5xhlKSdtWhB3FYGO20OxQYHmvsgt0vDV3UMbIP3S6MwGW52tLFhHW52oDTIKAi7sQWLSgN63M8R00KN5R6w1on6RWmxIb2BPeuDfSf1YVwZCclyfuoLiz4Nlb7%2F8ISqaMgf0AFnw%3D%3D&gui=q1ZKzs8tyM9LzSvxS8xNVbJSSk4sTk5MSQ3LTC1X0lHyTS3OCEotVrIy0DOE84IS89KBSqMN9AwMdYxidZSCElMyS4GKjA30zI1hXJgqIwMdE4PYWgA%3D)):

```javascript
let holeRadius = Slider("Radius", 30 , 20 , 40);

let sphere     = Sphere(50);
let cylinderZ  =                     Cylinder(holeRadius, 200, true);
let cylinderY  = Rotate([0,1,0], 90, Cylinder(holeRadius, 200, true));
let cylinderX  = Rotate([1,0,0], 90, Cylinder(holeRadius, 200, true));

Translate([0, 0, 50], Difference(sphere, [cylinderX, cylinderY, cylinderZ]));

let anExplorer = new oc.TopExp_Explorer(sceneShapes[0], oc.TopAbs_SHELL);

anExplorer.Init(sceneShapes[0], oc.TopAbs_SHELL);
const myShell = oc.TopoDS.prototype.Shell(anExplorer.Current()); // I think, casting to a Shell makes no difference
console.log("myShell before 'Next' ", myShell.ptr, myShell.IsNull());
anExplorer.Next();
console.log("myShell after 'Next' ", myShell.ptr, myShell.IsNull())
```

Output:
```
myShell before 'Next'  23878428 false
myShell after 'Next'  23878428 true
```

So, it seems like myShell is some kind of reference (or pointer) whose referenced object changes when you call `anExplorer.Next()`. The fact that the value of `ptr` doesn't change seems to mean that the each shell that is made available by `TopExp_Explorer` is created in the same object / point in memory.

I think what you want to do, in order for the `.push(myShell)` function to work, is to create a (shallow) copy of `myShell` before calling `anExplorer.Next()`. I have used this (hacky) approach in the examples repository. It's not pretty but it [works](https://zalo.github.io/CascadeStudio/?code=dZJNa8MwDIbv%2FRWiJweMm330FHYYW2GDboeltzFKlqpLwNhGdljH2H%2BfnKRp9hFjDJbe97ElW2OAymp8KnZ14%2BEKcl3vkMS8C8wlXKQg4Twul2mSzTQbvKuQEOJgQ7sRy5jcN6YMtTXwhiGvUGvhq8Jh8jmL2mgtzOrgtCUkthp8B1uqjXUc3B4TnUf2metXv83vVus14yNkbwnEiaLuTR0mHaPj1AOv4mfoEQ9BJAl014tjsSAMDZkeZW9z5cgGGz4cqq6ikf%2BmIUITEdlAKK3xAUrOtHKu8j%2FDbz134iifaMpA%2FFvmNNZHwxlDN1QYr4uA4jmVqVymL3I4c%2BTyJRrMYzO9co2vRAfoFV%2FZjOeIzODTQ7ffgJXf&gui=q1ZKzs8tyM9LzSvxS8xNVbJSSk4sTk5MSQ3LTC1X0lHyTS3OCEotVrIy0DOE84IS89KBSqMN9AwMdYxidZSCElMyS4GKjA30zI1hXJgqIwMdE4PYWgA%3D):
```javascript
let holeRadius = Slider("Radius", 30 , 20 , 40);
let sphere     = Sphere(50);
function getShell(shape){
    let anExplorer = new oc.TopExp_Explorer(shape, oc.TopAbs_SHELL);
    for (anExplorer.Init(shape, oc.TopAbs_SHELL); anExplorer.More(); anExplorer.Next()) {
        const currShell = anExplorer.Current();
        const newShell = new oc.TopExp_Explorer(currShell, oc.TopAbs_SHELL).Current();
        const shell1 = Translate([0,0,50], newShell);
        sceneShapes.push(shell1);
    };
};
const shell = getShell(sphere);
```

zalo, 2020-08-10:
That’s a useful explanation and a very clever workaround (akin to wishing for more wishes!)

I’ll poke around OpenCascade’s API to see if there are any more sinless methods of getting to the “bottom pointer”, or creating separate references, and see if I can make a wrapper for that.

zalo, 2020-08-10:
On a hunch, I exposed the so-called `Move` constructors in `TopoDS_Shape` (and all its faux "subclasses"), which (it turns out) allow you to steal the shape from the iterator (without copying it(?)) in a similar manner to the double `TopExp` example.

[You can see how that works here.](https://zalo.github.io/CascadeStudio/?code=fZHBS8MwFMbv%2FSseO6VQ2lTdaewgbqAwPZiBB5FSuzdbDElJUjYQ%2F3dfmtl2Aw0hh5f3%2Fb6XLxId1Fric7lrOgtLELLZoWGzUJglcM0hgSt%2F3PB4EUkS2LZGg%2BDXEramVFaWDtkrT3gy528JiL6BzXlMin2nKtdoBR%2FoRI1SMluXLcZfkQd4XqnWx1Zqg4Z4Cg%2Bgq3SrWyoWvxdBk5xubt9tIe7Xmw3hPWSvDbCRkj6oxv2pmNilj3Sy89ITHh2LYwjj%2BVVpZR1UnTH9%2BDTjpP2OyqhIsRj6syworO%2FOzyPKcx%2BSz2jgkX2WwUqjBaUdHLT5vLD%2BHzQGpleiCAmP8EB%2FIagdqLZChcLHY9O2szULBqcnfC8i2hNnMh6%2Frv9Z6vwB&gui=q1ZKzs8tyM9LzSvxS8xNVbJSSk4sTk5MSQ3LTC1X0lHyTS3OCEotVrIy0DOE84IS89KBSqMN9AwMdYxidZSCElMyS4GKjA30zI1hXJgqIwMdE4PYWgA%3D)

Let me know if this properly addresses your issue 👍

zalo, 2020-08-12:
I've used the move constructor to fix a problem with multiple .step import (which now works!) and to fix this problem, so I think it's safe to close this issue.

closed 2020-08-12 by zalo