zalo/CascadeStudio #6: Why GPL license?

fponticelli, 2020-08-12:
Disclaimer: Not a lawyer or an expert on licenses.

This is a really interesting project that could be a great alternative to OpenScad and others. I am wondering if the current licensing is deterring people from collaborating to it as it is not the most liberal license out there. What is the rationale to not use more permissive licenses like MIT or Apache 2?

Thanks for the great work!

zalo, 2020-08-12:
I'm open to moving to one of those (pretty much all of my other projects are); however I'm intentionally keeping the license restrictive and the project semi-secretive for now while I work towards what I consider a "V1 Feature Set" and still have "sole ownership" over the direction of the project.

I'd be kind of bummed if someone ran off with a fork and did a big publicity push before I had the chance to build out the infrastructure for a community (ala Shadertoy?)  This is my ultimate #1 priority with this project. :-)  I'll be happy to go MIT afterwards.

closed 2020-08-12 by zalo

fponticelli, 2020-08-12:
Thank you for the clarification!

zalo, 2020-09-15:
The gallery front/backend is taking longer than I hoped, so I went ahead and MIT Licensed it tonight.  This also includes a refactor, the addition of a bunch of new functionality, caching of intermediate shapes (to accelerate incremental builds), and the moving of all the CAD to a web worker (so the main thread isn’t frozen during lengthy computations).

I’m still hoping to get the gallery finalized, but it’s a bigger task than I had anticipated...

fponticelli, 2020-09-15:
That's great, thanks!