zalo/CascadeStudio #14: Selectors

parnold-x, 2020-09-19:
Hi, very cool Project!
Don't know if you know the CadQuery framework, but they have very powerful selectors. Maybe they would also fit for CascadeStudio. 
[https://cadquery.readthedocs.io/en/latest/selectors.html#combining-selectors](https://cadquery.readthedocs.io/en/latest/selectors.html#combining-selectorsurl)

zalo, 2020-09-19:
I just learned about this framework yesterday; it does several things really well.

Selector queries are an interesting alternative to the mouse-hover index selection I’m using right now.  Perhaps it would be possible to have a query selector function that outputs a list of array indices to retain the usage of both paradigms...

My soft plans for the mouse hover system are to allow clicking to insert indices directly into the index arrays (for fast selection of tricky to query  edges/faces).

I could imagine clicks also generating raycast queries which allow for persistent graphical selection of edges/faces/vertices...

zalo, 2020-10-01:
Appending @sgall17a 's thoughts on introspection from #23 here for spatial locality:
>On Hover in the GUI you get an index to a line or a face. Presumably this index refers to something that has gone through many transformations since starting as a primitive shape and thus is hard to relate to the original drawing.
>
>Also I presume that even if we noted the index from the GUI and could do some introspection (on say face[8] for instance) the index is likely to change after even simple modifications.
>
>It seems to me that if you can do fillet and chamfer programatically then should you know a fair bit about position of elements after various transformations.
>
>Would it be possible do some introspection with this mechanism.
>
>For instance say we had a box and knew the absolute coordinates of some point on the box like a corner. We could define this point then at the point of visualisation echo its new absolute position. A point can move or disappear (like in difference operation). If we knew the point survived transformations because it was still visible after being transformed then knowing its absolute position through code would be of great benefit.

whosawhatsis, 2020-10-06:
The system where fillets and chamfers need to be selected manually really makes it impossible to use them with programmatically-generated models. If I had an element in my model with a variable number of some feature like sides or holes, I would have to manually identify all of the resulting edges that need fillets after changing that variable.

A better system would be to make fillet an option on boolean operations, so that any new edges created by that operation (usually concave edges for a union or intersection, and concave edges for a difference) will have a fillet applied.

Another option would be to treat fillet like boolean operation. The shape you want to apply fillets to is the first object, and you create a second shape to use like a mask for the fillet operation, so that any edges of the first object that fall entirely within the volume of the second object will have the fillet applied, to them.

These two systems aren't mutually exclusive, and would actually complement one another really well. They're also both very automatable, and applicable to a wide range of possible shapes.

parnold-x, 2020-10-06:
I think these approaches are way to powerless. The selector approach in CADquery proved to be very powerful and convenient

tingspain, 2021-02-16:
@parnold-x have you done any progress or implementation in this regard? I think it is a very powerful feature.

parnold-x, 2021-02-16:
@tingspain Nope, I am not really familiar with web stuff.