zalo/CascadeStudio #16: 3D Textures

fponticelli, 2020-09-23:
Maybe more of a question than an issue since I am not very familiar with the OpenCascade API. Is there any way to apply 3D Textures to surfaces? Something like what is [described here](https://www.cadimensions.com/blog/how-to-make-stunning-3d-prints-with-3d-texture-in-solidworks-2019/).
I think such feature would make this library standout from many others (certainly OpenSCAD).
If OpenCascade doesn't support it naively, are there post processing strategies that people are aware of?

zalo, 2020-09-24:
That is a really cool feature!  Unfortunately it doesn’t look like OpenCASCADE supports heightmap-based surface modifiers(?).

If we wanted to hack it in anyway, the first thing we’d want to do is generate UVs for each of the contiguous faces of the model (either using an off-the-shelf UV unwrapping system, or building one using OpenCASCADE’s own surface parameterizations (though I don’t think that will help at much over pure mesh systems)).

Then, we would add a displacement mapping module which would take in the Mesh+UVs and a series of heightmap textures to produce a new Mesh.  This is sort of already implemented in threejs:
https://threejs.org/examples/webgl_materials_displacementmap.html

Unfortunately, this would only be for mesh-based exports, so STEP files would be left out.

Speaking of which, do 3D Textured STEP files exported from SolidWorks import into CascadeStudio?  That might give me a better hint as to how to implement them...

fponticelli, 2020-09-24:
The three.js example applies the modifiers in the GPU so you wouldn't be able to export the geometry to STL, right? I don't have access to Solidworks but I just asked the author of the article if he would be able to provide an example. Fingers crossed :)

zalo, 2020-10-02:
I’m going to close this issue for now, since it seems like this will have a hard time being implemented purely in the OpenCASCADE Kernel (and I’m a bit worried about Mesh Post Processes since those are better suited for MeshMixer, Cotangent, or OpenSCAD itself).

Mesh UVs aren’t totally out of scope, however, and may be an important intermediate step towards implementing this functionality in post.  So I’ll consider spending some time at some point figuring out the best way to implement those...

I’d also be happy to examine any example STEP or IGES files that you glean from the SolidWorks implementation!

closed 2020-10-02 by zalo

zalo, 2021-03-30:
Automatic Mesh UV unwrapping has been implemented, so it’s possible to subdivide, tessellate, and use a displacement map in Blender for this effect now.

Maker’s Muse also just published a video on direct-to-GCode textured surfaces, which should theoretically be compatible with CascadeStudio output parts:
https://youtu.be/eFrSeppuGZg

fponticelli, 2021-03-30:
Thank you for following up.