zalo/CascadeStudio #21: Feature request: Make project JSON more friendly for software version control

udif, 2020-09-30:
At the moment, the script is saved as a single string with embedded newlines.
In order to make diffing different versions easier, could you please store the multiline script as a string array?
inspired by: https://stackoverflow.com/a/7744658
e.g. instead of :

        "code": "// a\n// b\n// c\n",

It would be stored as :

        "code": [
                "// a",
                "// b",
                "// c"
        ],

You could still support reading single string scripts, but write it out as a string array.

zalo, 2020-09-30:
This is a great idea! Will definitely investigate 👍

zalo, 2020-10-02:
Alrighty; it's implemented and merged.

Pre-multiline serialized projects will automatically be converted to multiline projects upon re-saving 👍 

Thank you for the suggestion!

closed 2020-10-02 by zalo