zalo/CascadeStudio #22: Questions about syntax

sgall17a, 2020-09-30:
This seems a very promising progression for OpenScad.  
I have been playing around with it and I have few questions

This is some test code I am using for examples.
```
 let ball = Sphere(10)
//let ball = Cylinder(20,50)

let bolero = Translate([50,0,0], ball); 

function draw(shape){

 [1,2,3,4,5,6].forEach(x=>{
    console.log(x)
    Rotate([0,0,1],60*x,shape)})
}
draw(bolero)
```

Questions.
1.  If my Array is [1,2,3,4,5,6] I get 6 balls.   However if my array is [0,1,2,3,4,5] I only get 5 balls. I think I should get 6 since 5*60 = 360 = 0 degrees.  The zero degree case does not seem to show.

2.  If I comment out ball then it persists between evaluations. (I can see I should reset the project).  Not really a question I suppose.t

3. A. If I just write Sphere(40) it draws a sphere.  Fair enough.
B. If I write let sp = Sphere(40) it also draws a sphere.  Why? Shouldn't it just assign Sphere to sp.
C.  If I write let sp = Sphere(40) as above, but also pass it as a parameter to my function draw(shape), it does NOT draw the original sphere.  I think this is the correct behaviour but it seems inconsistent with B.

COMMENT AND QUESTIONS. (probably ill-informed but I am trying to understand this stuff).
Looking  at the code it seems that shapes are drawn on creation.
eg   sceneShapes.push(curBox);
But why does it not draw when passed as a function?  Is creation delayed somehow?

I dont think shapes should be drawn on creation.  I think that is a mistake that OpenScad makes, too.
Like any mathematical calculation the print should not occur during the calculation phase, only when we ask it to print (or draw or whatever the appropriate name is in this context).

This way the work flow would be to design our shapes in a modular way and with composition.  Drawing should be only done explicitly when we have finished designing.  This would also allow a sort of backdoor to introspection because we could introspect bits of code (by a special function say boundingbox for instance) separate from the drawing process.

zalo, 2020-10-01:
1. This is actually [the result of a bug on this line](https://github.com/zalo/CascadeStudio/blob/master/js/CADWorker/CascadeStudioStandardLibrary.js#L374); I'll look into fixing this (and related transformation bugs).

2. Yeah, the javascript variable scope is persisted over time.  I'll consider the implications of that, however, I'm about to make use of it in the Typescript PR to persistently import libraries.

3. Objects are only drawn if they appear within the `sceneShapes` array by the time your script finishes.  You can explicitly control what draws by clearing the `sceneShapes` array, and then manually adding what you like to it.

Individual shapes/solids each have their own indexing system, so they shouldn't interfere with each other when it comes to hover introspection, but I agree this can be made better.

zalo, 2020-10-02:
Alrighty, I just fixed the rotation inconsistency.

I appreciate the drawing side-effects in OpenSCAD (and thus CascadeStudio) because they assist with learning.    Since `You can explicitly control what draws by clearing the sceneShapes array, and then manually adding what you like to it.`, I'm going to mark this issue closed for now.

However, feel free to ask more questions here (it still creates notifications).

closed 2020-10-02 by zalo