zalo/CascadeStudio #25: Units and aggressive fog

codri, 2020-10-03:
Hi,

I assume the system is made for arbitrary units, like openscad.
When I assume the units to be mm, and I make a 15mmx15mmx400mm prism the fog hides most of the model.
If I try and scale the units down, ex: instead of passing  15, 400, I'm passing 1.4 and 40, I get this error:

```
Line 140: Uncaught RuntimeError: INTERNAL OPENCASCADE ERROR DURING GENERATE: function signature mismatch
```
It looks like not all the functions accept both floats and ints, or it's not possible to mid argument types.

Is this intentional? I see that the fog is added in the three.js scene, I assume it's some sort of a standard scene type.
Maybe it's worth allowing adjusting the fog intensity? Or add the ability to toggle it?

Thanks

zalo, 2020-10-03:
It should be possible to pass floats into most inputs; it might be necessary to disable caching, refresh the page, or open it in a new tab, however (some backend instabilities still remain).  Can you paste a link to the model that you find to give you the error?

I’ll modify the fog based on the bounding box of the model so it’s always visible.

codri, 2020-10-03:
Yep. Here's the model: https://zalo.github.io/CascadeStudio/?code=zVZbb5swFH7PrziPuCUE0kRb1WUP624Pe4jatXtAPDBwiiVqd4aEtVP%2F%2B46xMQY13aZKU1AExj6X73znQjLBqxouaFbDCrzGh4LA6i2sRXl%2FI7gXx6EP6pf4EDeDZWGWoVkm5GySaWsiV8ZkZ%2Bz8vmQ8p1LvdFJrdkeNGLOiH37WcptT7z3bbKikPKPeOZNZST1J0JdZM0kSol0r78rzZLPlWc0Eh6tPUogdfUdTyfiNxy7SnG0rH0S3aAj8mgBeJa3hphVGHF9lyqsyrakX95Iwg7kO03huZvOjMDglxBqoqHa7Agdze6oux6oFYjhUnHvGFUyB9fCIb%2FVjjS9pN1SY6ilpvZUcDexEuaOeQeBD9DrUpETKA0o%2FOrRcD2npOJjNYCD2hXGUcKXc03WZVjXLnhcyOczXUmwYcla4fGeU11QO%2BZ4ix9O5w8rChwVSbJVylmIxpmWFenGi9zdCertUAsO96Awfb2CJj%2BPjzttYGeUuRK0c2uOWYV1EUeIPtqcegyM4DQkcw2I5PHOhh8F8aQg34MMAN16ZCmnTZlc2juBuWxVe96oFHvt4MyH5mCTFSevFWnuq3tTVokA652QIO%2B7hRcGCGLi4RLPBkiQjwMi%2FBYRk3%2B7P2RVnalRYfZ3j3nkQBDbyflcH2b4mbrILUdI20VbS9F7kxOMgOVEBtTenT8PgxGRAh%2BX2jcObisvXHsmghi9%2FbFNJ1YjyKvZAsSvTsrxsVyXlXYkZg93Q%2BrvMaHvqPs6PE1TnTg%2Bg4ZutNGUDJ0cPbbRB3Jw6M0VXPMZhxsog8o8yvaXmQ6Cj1OP6Drn4xvK6UP22HB%2BgR9yfT7qZck0lzglsuu%2BprCajjPWfFIdma9%2B3FhUGMlZuXqKMms1LPP%2BDcsfEZyHZg%2BD1Hi4c%2FdDvSXbafN%2FQMjPnNBxW0Z%2BwNVgg9mA8pAzsp%2FE1B46v%2FStwmLwdDi7bAc%2FjiWyX%2FRc8hat0MNiaA%2BOqOSCu8JOhvxTLEB3gjfwG&gui=PYuxDkBAEAX%2F5dUbWUd1jUJNcYVGFJuzQeGIIwrx7zR0M8nMBb%2FM6xI07LXMCgsv0UuvzaQnCJXG0WmETZPc%2FOokDG%2FbcsIpmY5Qih%2B1gN23QwlO%2Bul4n4w%2F%2FgbDlHN3Pw%3D%3D

codri, 2020-10-03:
I'm trying to understand if I'm doing something wrong. In openscad the units are arbitrary(within the float value range), I used to choose 1mm as the base unit.
I assumed it was the same here. But the fog it makes it seem like there is a limit on size. In the example there is a 400mmx400mm frame.

Great project btw, it fixes almost all the problems I had with openscad, and it strongly makes me abandon my own in progress lua based openscad version. I was planning to add PBR materials, maybe native bom support, all in a lua DSL that renders natively, all a whole bunch of other plans.

But we'll see, maybe I'll steal your viewer style, since Phong shadings seems good enough for now along with the nice plane and maybe even the fog.

zalo, 2020-10-04:
You've done nothing wrong 👍 ; I've only used Cascade Studio for small 3D printing projects, so I never ran into any of the issues associated with dimensionally large objects.   I just added some logic that will push the fog back according to the bounding box of the exported parts, so the part in your link can be seen clearly now.  I'm also not seeing any issues with floating-point numbers, so I assume it was just a bad session state.

For what it's worth, CascadeStudio uses a [Matcap Material](https://threejs.org/docs/#api/en/materials/MeshMatcapMaterial) to do its shading.   It's a cheap trick where you simply sample [a texture](https://github.com/zalo/CascadeStudio/blob/master/textures/dullFrontLitMetal.png) according to a fragment's screen-space normals (without any regard to lighting calculations).  Someday it would be nice to blend it with [an Envmap](https://threejs.org/examples/#webgl_materials_cubemap_balls_reflection) and add some [SSAO](https://threejs.org/examples/?q=SSAO#webgl_postprocessing_ssao) to achieve a prettier preview (ala Fusion 360).  Heck, maybe we'll even get WebRTX at some point xD

closed 2020-10-04 by zalo