zalo/CascadeStudio #26: Orientation cube in the viewer

codri, 2020-10-03:
It would be great to have a classic orientation cube in the viewer, that would indicate the camera orientation,
as well as allow for easy navigation between the views.

[image]

Is this a feature you think would be handy for the project?

Thanks

zalo, 2020-10-04:
This is a good idea; I'll have to think about the best way to implement it.

zalo, 2020-10-04:
Did [a test implementation](https://github.com/zalo/CascadeStudio/pull/27) that you can try [here](https://cascade-studio-git-feat-view-cube.zalo.vercel.app/), and after considering the added complexity:functionality trade-off, I might just split the difference and render a non-interactive orthonormal basis as a simple reference for scripting coordinate calculations.

codri, 2020-10-04:
That was quick. Looks good. The size is a bit large, and the perspective looks a bit odd, probably because of the vertical distiortion.

Also, I would suggest using the labels (-x -> front, x -> back, -y -> left, y -> right, +z -> up, -z -> down) or something similar.
Maybe also worth having the axes along the edges.
[image]

But overall looks good. Be careful not to burn out with work over the weekend(just a suggestion, you know better what you're capable of).

johnsonm, 2020-10-05:
What would you think of having the ability to instead just have X, Y, and Z arrows? I'm used to that in OpenSCAD, FreeCAD, SolidWorks, and IIRC all three put that indicator in a corner; lower left for OpenSCAD and lower right for FreeCAD.

Here's what it looks like in FreeCAD: 
[image]

codri, 2020-10-06:
One of the main advantages of the cube is being able to click on a face,
and it would position the camera on top of that plane, thus allowing a faster switch to a 2d view.
But this can also be done with keyboard shortcuts, so I don't know, but I don't have any strong opinions on these.

johnsonm, 2020-10-07:
Ah, I'm used to that being a different tool.

Again the FreeCAD version. First, showing a 45⁰ from every axis view:

[image]

Then a front view after clicking on "Front":

[image]

You can see that it also incorporates the axis hint graphic.

All the light arrows and tabs at the edges of the cube are clickable to rotate around the 6DOF in 45⁰ increments. At least in the FreeCAD version, the cube can't be directly manipulated, just clicked through standard orientations. I keep trying to grab it to rotate it and instead snapping to a standard view by accident. ☺

brad-jones, 2020-12-09:
> But this can also be done with keyboard shortcuts
@codri care to share what the keyboard shortcuts are?

zalo, 2020-12-10:
I don't believe I've implemented view orientation keyboard shortcuts 🙃 

I assume, if implemented, the function would be something like a combination of [the rotation code from here](https://github.com/zalo/CascadeStudio/pull/27/files#diff-734677b08737cfcfba04b154c9b8955d80d292748b3f1d8bc0de95007472fd8aR371-R398) and [the keyboard shortcuts from here](https://github.com/zalo/CascadeStudio/blob/master/js/MainPage/CascadeViewHandles.js#L92-L121).

borioan, 2020-12-10:
I love the idea of using an orientation cube in the viewer!!! It is perfect and rapid option when you need a 2D view of a specific plane (front for example). Can a version be implemented based on an orthographic camera also?