zalo/CascadeStudio #28: Line 18: Uncaught Null Shape detected in sceneShapes; skipping

johnsonm, 2020-10-05:
I'm trying our CascadeStudio after years of OpenSCAD and there are a lot of things I really like about it! A well-known language, fillets and chamfers, and so far zero Z-fighting come to mind. Thanks for working on it, it's really an attractive idea. I looked around for something geometrical to model as an exercise and saw a game, so tried modeling it as an exercise to learn CascadeStudio.

The line numbers in the error messages not being linked back to the line numbers in the source has given me a little trouble finding the occasional misplace parenthesis. But this looks like even if the model has a bug, it _might_ also be a bug in CascadeStudio. I'm getting lots of errors of this sort:

```
Line 18: Uncaught Null Shape detected in sceneShapes; skipping: {\"ptr\":16073848,\"hash\":-1935030109}
...
```

I narrowed my model down to something relatively simple:

https://zalo.github.io/CascadeStudio/?code=hVJNa4QwEL37K%2BaoNf3QuqVg97KnFhYKpaWHsgfrjq4gsWgEbdn%2F3kliPnZ7aBCTGTPvvXljiwLKFou%2B4CXCGpI8aCn1WQy4RU6JdKUzI2%2FEQPG9DvfvzZ4ic%2B8a0qssyZZi%2FMb2iQ90XMPtkuuKXtaIA%2BU01oWtjiGVkS3LA1fziE19kECLji%2BsberOpp6rStO5zySJgD0MQq1GXoqm4%2FJaGMFPALQWiC3WVK%2F6unSW5OpKj2LsObx0ohAYftwwoCfZMchWDDbdFOp6Bt6uSRmIfsQoyoOjRy8b30hlJyKUVtLwSsxD6zH5RlBbO83pLGXw53zOLjmqrodQEk1q0LQ96FHQMY6NEqNmsp6GEzmiZ0bskRtcbgss9KyhZwc9n0Ib%2BNnBz%2F%2FCy2XceePkoG6e%2BVYteplBZu6%2FIMP0xI0Tch0D917mq0DVpLwB5b8%3D&gui=Pc0xC8IwEAXgvyI3h3AmLmYR0VWHDC7icCanDTZp6bU4iP%2FdKNTxPd7He0Hoct8VLuORMoODQBIo8inxExQcWBrPAg61sf%2Foqdzr9owal8pcFOwoNLwBNw4TK%2FAU01SNtXo9p5kYVCusYivC%2BdpyBHejVqra0%2FBY9InD9%2B7XvT8%3D

Thanks!

cpu-1, 2020-10-05:
I changed to an array for:

function Union(objectsToJoin: oc.TopoDS_Shape[], keepObjects?: boolean, fuzzValue?: number, keepEdges?: boolean): oc.TopoDS_Shape

Also nudged the pegs into the board by 1 unit as co-planar faces can slow down boolean operations.

let clearance = 1;
let baseLen = 25;
let units = 8;
let dWid = baseLen / 2.414;
let bezelInset = 3;
let boardWidth = units * baseLen + 2 * bezelInset;

let boardHeight = 5;
let pegHeight = 6;
let pegOffset = pegHeight / 2 + boardHeight;

function peg() {
    let pegLeg = dWid - clearance;
    return Rotate([0, 0, 1], 45, Box(pegLeg, pegLeg, pegHeight, true));
}
var a=[]
function baseBoard(){
    let board = Translate([0, 0, boardHeight / 2], Box(boardWidth, boardWidth, boardHeight, true));
    for (let x = 1; x < units; x++) {
        let xOffset = (x - units / 2) * baseLen;
        for (let y = 1; y < units; y++) {
            let yOffset = (y - units / 2) * baseLen;
            //board = Union(board, Translate([xOffset, yOffset, pegOffset],  peg()));
            a.push(  Translate([xOffset, yOffset, pegOffset-1],  peg())  )
        }
    }
    a.push(board);    board=Union(a);
    return board;
}

    
baseBoard();

cpu-1, 2020-10-05:
With formatting:

https://zalo.github.io/CascadeStudio/?code=jZJBS8MwFMfv%2BRTv2Lhss3MTofayk8JAEMXD6CFu6VYo6WhTaZV9d1%2BSJun0oKG0yWve7%2F3zfymFgl0peM3lTkAKcUJKDL3zRmyExMBiZSOtLFSD6zu73L8Ve1y5fXNYzJbxckgWn6J8lA1OU7gZYhWvdY46Ysyyrnz2BBZ65dMSEnIeRHE4atCg4yQOPnTrQ095bsuF3ygJwSMGUvNW7lRRSb0tovBFAMeA2IgD5ptzTYMlidlSC9XWEp4rxZWIttcM8IkzBssVg3XVRTafwehrizJQdSsoTciZfPAaeLrNggxtwForjGjQYiSjlBcU0JSjgmM%2F8HSZLR2cZfBr%2FlOErpFXNUS6UGf6jZ972xGcTibOFaem89ZGHRpjW4fVaehf4hM8urfoPqD7S7TD9wHf%2F4nXYz53%2FrxKtNAen43NGhQzx2bhgqBltvXOCzf47NQ2xwj%2ByZnGgQRAPepMwntAGn000REzTa1sTi9ulvml74gJktG9SMg3&gui=Pc0xC8IwEAXgvyI3h3AmLmYR0VWHDC7icCanDTZp6bU4iP%2FdKNTxPd7He0Hoct8VLuORMoODQBIo8inxExQcWBrPAg61sf%2Foqdzr9owal8pcFOwoNLwBNw4TK%2FAU01SNtXo9p5kYVCusYivC%2BdpyBHejVqra0%2FBY9InD9%2B7XvT8%3D#

johnsonm, 2020-10-05:
Thank you! This was excerpted from a larger model. I feel silly about missing the list on the `Union` invocation; every other `Union` in my surrounding code not copied when I made this simplified model did that right.

All of my OpenSCAD models have explicit overlap to avoid co-planar surfaces; I just was noticing that you managed to avoid Z-fighting in some trivial examples, and I was playing with co-planar surfaces on purpose. ☺

Here's my version — which, since I sometimes have thin parts, makes me ask how much overlap is useful for avoiding co-planar surfaces. Do you have a heuristic similarity value which should be a minimum overlap for computational efficiency?

```javascript
function peg() {
    let pegLeg = dWid - clearance;
    return Rotate([0, 0, 1], 45, Box(pegLeg, pegLeg, pegHeight+0.2, true));
}

function baseBoard() {

    let board = [Translate([0, 0, boardHeight / 2], Box(boardWidth, boardWidth, boardHeight, true))];
    for (let x = 1; x < units; x++) {
        let xOffset = (x - units / 2) * baseLen;
        for (let y = 1; y < units; y++) {
            let yOffset = (y - units / 2) * baseLen;
            board.push(Translate([xOffset, yOffset, pegOffset-0.1],  peg()));
        }
    }
    return Union(board);
}
```

johnsonm, 2020-10-05:
Here's my completed [test model](https://zalo.github.io/CascadeStudio/?code=pVZLb9pAEL7zK1Y52WEB40KVKOXQJK0aKVKjplUPiIOBMVhxbGSbBlrx3zv78O6sTR5VrQSvx7PfvL6ZdQoVW6QQFVG2ADZhw4tOiqJNVFRXRDxW4nlUwi1kKAi1ZJslVYnPZ%2Bpx%2BTNZ4lOtN2BhfzQc6c3wG9KbrMTlhL3TsjwqxJ5qjTKFdWp2d1konsy2i47Fqfc44F8gWa0F%2BjAkUoRaSV1ijEArQa9ly3X5axwrx8PaCwFmDOpwNrAyovdGZPba15gY9IFg1Gl%2FygyA1e6ykbYqFMo7KO6TJejopTwqS3icpyCSf7WGxcM833knH2vpCWdxlJbg6ypFxcNdAgsoHfVrFLONlJMNnXibLaokz6Rxz2d%2FOgyv2ptbWJGC9yyZLoyaeHmHqqh3iXb0Ls6chQqVs6rYCqtibwHVtsjYTVZBUYL0wZvKN%2BKqUbmRfMurqAJvGnCGf8MZZ6MxN3q%2B1Jsh9oHGBCs3JFipiCSTW%2BFol45bksHJ%2FZyRu4qsG%2FRDHV3DBeHgpWCC44jkBvox%2FY7my5SYo8xDHs2UYUtuzlprJ7n%2BTMUS5wXzhKmd7Hu8fVAtiMtut%2Fal9mdnWOztMC%2BqV9G8bxv2wmww0HsFvbfQexe6ht9b%2BP2r8LL8Iq7%2BZluuPZIh7SWv8bjtv17QF4VSFfd9i3bo2F9d3x%2BZ4Jo00SyWmAimUFq9XSI7WGSFjK3PSYrRflquoPSccK6TOIYCkGiuXFxX6%2BgRX6pdstZ2rHE6%2BahhXWzuDrmgHww5m%2BL%2FeObzlqUG2Qh3esTMIGxGiMhkuhNSOo5SJlp1w0rHm4Zzw%2F5YuI0%2Fw3OrqFr6WIXuoWoWqd2z53hXlbZTxU1Az6PnihraztE4CDHDwYzXvCBuN1L5Go4Lo2dVc1jhGKORmRNBzIlmV4vWC1Tr0VNjELZ6UL5utVJA58hAllj1oVzvsS89e1A7weiUqOOCdtpxQ8%2BgvGT%2Bn60f6W7pjEpwEjPPnKB1Zo43NfkYmHE6un1y3mHmX0hk121J2glOkyF%2BzdmhdCDQnK1J4jcqnk2w3hnWe4Q3t8TCpeMNMHeO3EPjZBCIz58LlmuT4LUhPxiwcp0%2FsWodVQx%2BQbFnm7xMJK8X%2BTZdYlgsFuNxyZ4SzEyeActj1AcWnrFoUW2jVDHIwSWJ7mlGnEovVKYNZ8z6dId9%2FTbNfbvoXTLFLMvah8mBAX4%2BMcEs%2B7V1hFrUbIP%2BgSWybHpthQbsESYdaQMyU2oE7Zfyg9C3hd2k7H%2F6%2BWwDvNHtvw%3D%3D&gui=Pc2xDsIgFIXhVzF3Jg2CiyzG6FoHBhfjcIWjJRbalDYOxncXBxz%2F5Hw5b3JDHIeENJ84ggw5zo49zgEvEtQidxaZjGyU%2Fqfl9Cjbi2zkWqiroAO7Djsy87RAkGUflmK0bra1KlFSbGQR%2B5wRbz18RUeenqsxwP3e7txnfL4%3D](https://zalo.github.io/CascadeStudio/?code=pVZLb9pAEL7zK1Y52WEB40KVKOXQJK0aKVKjplUPiIOBMVhxbGSbBlrx3zv78O6sTR5VrQSvx7PfvL6ZdQoVW6QQFVG2ADZhw4tOiqJNVFRXRDxW4nlUwi1kKAi1ZJslVYnPZ%2Bpx%2BTNZ4lOtN2BhfzQc6c3wG9KbrMTlhL3TsjwqxJ5qjTKFdWp2d1konsy2i47Fqfc44F8gWa0F%2BjAkUoRaSV1ijEArQa9ly3X5axwrx8PaCwFmDOpwNrAyovdGZPba15gY9IFg1Gl%2FygyA1e6ykbYqFMo7KO6TJejopTwqS3icpyCSf7WGxcM833knH2vpCWdxlJbg6ypFxcNdAgsoHfVrFLONlJMNnXibLaokz6Rxz2d%2FOgyv2ptbWJGC9yyZLoyaeHmHqqh3iXb0Ls6chQqVs6rYCqtibwHVtsjYTVZBUYL0wZvKN%2BKqUbmRfMurqAJvGnCGf8MZZ6MxN3q%2B1Jsh9oHGBCs3JFipiCSTW%2BFol45bksHJ%2FZyRu4qsG%2FRDHV3DBeHgpWCC44jkBvox%2FY7my5SYo8xDHs2UYUtuzlprJ7n%2BTMUS5wXzhKmd7Hu8fVAtiMtut%2Fal9mdnWOztMC%2BqV9G8bxv2wmww0HsFvbfQexe6ht9b%2BP2r8LL8Iq7%2BZluuPZIh7SWv8bjtv17QF4VSFfd9i3bo2F9d3x%2BZ4Jo00SyWmAimUFq9XSI7WGSFjK3PSYrRflquoPSccK6TOIYCkGiuXFxX6%2BgRX6pdstZ2rHE6%2BahhXWzuDrmgHww5m%2BL%2FeObzlqUG2Qh3esTMIGxGiMhkuhNSOo5SJlp1w0rHm4Zzw%2F5YuI0%2Fw3OrqFr6WIXuoWoWqd2z53hXlbZTxU1Az6PnihraztE4CDHDwYzXvCBuN1L5Go4Lo2dVc1jhGKORmRNBzIlmV4vWC1Tr0VNjELZ6UL5utVJA58hAllj1oVzvsS89e1A7weiUqOOCdtpxQ8%2BgvGT%2Bn60f6W7pjEpwEjPPnKB1Zo43NfkYmHE6un1y3mHmX0hk121J2glOkyF%2BzdmhdCDQnK1J4jcqnk2w3hnWe4Q3t8TCpeMNMHeO3EPjZBCIz58LlmuT4LUhPxiwcp0%2FsWodVQx%2BQbFnm7xMJK8X%2BTZdYlgsFuNxyZ4SzEyeActj1AcWnrFoUW2jVDHIwSWJ7mlGnEovVKYNZ8z6dId9%2FTbNfbvoXTLFLMvah8mBAX4%2BMcEs%2B7V1hFrUbIP%2BgSWybHpthQbsESYdaQMyU2oE7Zfyg9C3hd2k7H%2F6%2BWwDvNHtvw%3D%3D&gui=Pc2xDsIgFIXhVzF3Jg2CiyzG6FoHBhfjcIWjJRbalDYOxncXBxz%2F5Hw5b3JDHIeENJ84ggw5zo49zgEvEtQidxaZjGyU%2Fqfl9Cjbi2zkWqiroAO7Djsy87RAkGUflmK0bra1KlFSbGQR%2B5wRbz18RUeenqsxwP3e7txnfL4%3D) just in case it's interesting as an example. It's a representation of the "Pathagon" game that I saw when I looked around for something geometric to try to model as a real-life example.

```javascript
let clearance = 1;
let partClearance = 5;
let baseLen = 25;
let units = 8;
let dWid = baseLen / 2.414;
let bezelInset = 3;
let boardWidth = units * baseLen + 2 * bezelInset;

let bezelWidth = 14;
let bezelHeight = 12;
let bezelLength = boardWidth + 2 * bezelWidth - 2 * bezelInset;
let bezelInsetOffset = 2;

let boardHeight = 3;
let pegHeight = 6;
let pegOffset = pegHeight / 2 + boardHeight;
let pawnHeight = pegHeight + 4;

let pawnsPerSide = 14;

let assembled = Checkbox("Assembled", false);
let darkPieces = Checkbox("Dark pieces", false);

function pawn() {
    let pawnLeg = baseLen - clearance;
    let basePawn = Box(pawnLeg, pawnLeg, pawnHeight, true);
    return Intersection([
        basePawn,
        Rotate([0, 0, 1], 45, basePawn)
    ]);
}

function peg() {
    let pegLeg = dWid - clearance;
    return Rotate([0, 0, 1], 45, Box(pegLeg, pegLeg, pegHeight+0.2, true));
}

function baseBoard() {
    let board = [Translate([0, 0, boardHeight / 2], Box(boardWidth, boardWidth, boardHeight, true))];
    for (let x = 1; x < units; x++) {
        let xOffset = (x - units / 2) * baseLen;
        for (let y = 1; y < units; y++) {
            let yOffset = (y - units / 2) * baseLen;
            board.push(Translate([xOffset, yOffset, pegOffset-0.1],  peg()));
        }
    }
    return Union(board);
}

function bezel() {
    return Translate([0, 0, bezelWidth / 2],
        FilletEdges(
            Difference(
                ChamferEdges(Box(bezelLength, bezelHeight, bezelWidth, true), bezelWidth - 0.01, [1, 5]),
                [Translate([0, boardHeight-bezelHeight/2, bezelWidth / 2 - bezelInset / 2], Box(bezelLength, boardHeight, bezelInset, true))]
            ),
            1.5, [15, 19]
        )
    );
}

function bezelSet() {
    return Rotate([0, 0, 1], 90, Union([
        Translate([0, -(bezelHeight / 2 + partClearance/2), 0], bezel()),
        Translate([0, bezelHeight / 2 + partClearance/2, 0], bezel())
    ]));
}
[... 6 lines cut ...]
    }
    return Union(pawns);
}

if (assembled) {
    Translate([0, 0, bezelInsetOffset], baseBoard());
[... cut ...]
            Translate([-baseLen*units/2 + baseLen/2 + baseLen*x, -baseLen*units/2 + baseLen/2 + baseLen*y, bezelInsetOffset+boardHeight], pawn());
        }
    }
} else if (darkPieces) {
    Translate([baseLen/2 + partClearance, 0, 0], pawnSet());
    Translate([-(bezelWidth + partClearance), 0, 0], bezelSet());
} else {
    baseBoard();
    Translate([boardWidth/2 + baseLen/2 + partClearance, 0, 0], pawnSet());
    Translate([-(boardWidth/2 + bezelWidth + partClearance), 0, 0], bezelSet());
}
```

closed 2020-10-05 by johnsonm

cpu-1, 2020-10-06:
Thanks Michael!

It's an interesting example.