zalo/CascadeStudio #29: Add Color() for viewing?

johnsonm, 2020-10-05:
I could imagine `Color("colorspec", [oc.TopoDS_Shape])` or something like that to apply a color to a list of objects.

Sometimes I apply random colors to objects just to be able to tell them apart easily when I'm making a complex model.

zalo, 2020-10-08:
This is a good idea.  It'll be slightly tricky to implement since I recently refactored the drawing system to draw everything in a single draw-call (to make editing .STL Meshes feasible), but this may be worth adding the additional complexity...

It would be _really_ slick to propagate colors through changed faces on CSG operations, but that requires a level of OpenCASCADE access that is beyond my familiarity right now.

flolu, 2022-07-12:
@zalo You mentioned _"per-face highlighting"_. Would #147 be possible?