zalo/CascadeStudio #30: Feature Request: Better project saves

udif, 2020-10-15:
When saving a project, I face 2 problems causing inconvenience:
1. The save file is always the same, you cannot control it
2. The location is always the download folder

I would like to see an option for controlling the filename, perhaps by hinting it inside the project (a specially marked comment perhaps)
As for (2), I hardly know JS, and have zero knowledge about the security implication of file writes, so I don't know how easy it is to write a file to the local filesystem by the client, or pop up a 'saveAs' dialog, (and keeping the default directory from the last time it was saved).
I understand that for security reason, a client app cannot arbitrarily write to the client disk, but with a saveAs dialog, perhaps this is possible, since it requires user intervention?

zalo, 2020-10-30:
Unfortunately, proper Save File dialogs don't seem to be possible in modern browsers.   Filename hints are certainly possible; I'll investigate this as part of the set of rewrites coming in the future...

brad, 2020-12-05:
@zalo I'm looking into this for my personal sanity. I've got it generally working locally. Can you assign this issue to me and I'll make a PR once I clean it up?

zalo, 2021-01-09:
@brad 's PR has been merged, so I'm closing this issue.

closed 2021-01-09 by zalo