zalo/CascadeStudio #32: Problems with npm install

UriShX, 2020-10-15:
First of all, I found this project through Hackady, and I think it's really great! Thank you!

I tried to add a library \([newtype-ts](https://github.com/gcanti/newtype-ts), as described in PR #31\), but running `npm install newtype-ts` scrambles all my local `node-modules` libraries.

I think there should be a better / more robust way for adding, changing, or upgrading npm modules.

zalo, 2020-10-30:
Yes, there certainly should; the way this project is laid out currently fights npm and requires significant manual rejiggering of the `node_modules` folder.  At some point, I'm hoping to come back and refactor it to use entirely es5 modules and proper typescript (rather than global-namespace embeds and vanilla js).  This is going to need significant work to bring some of the dependencies in-line, so I don't currently have an estimate for when that's going to come about...

(Thank you for the PR, by the way!)

UriShX, 2020-10-31:
I need to learn how to set up [webpack](https://webpack.js.org/) for a different project. If it's all right with you, I could try and set up CascadeStudio to be bundled with Webpack, perhaps with Babel for TS support. I don't really know how to do it yet, but I'd love to give this a go.

zalo, 2020-11-02:
I was thinking about teaching myself the same thing a month ago, but that was when I learned that webpack is an absolute monstrosity, and probably the root of a lot of evil in modern web tech.   You can see the half-baked carnage from such experimentation here in [`feat-modules`](https://github.com/zalo/CascadeStudio/compare/feat-modules).

Once the trauma from that has worn off, my eventual hope is to come back and try [Snowpack](https://www.snowpack.dev/), which markets itself as a more humane webpack (that can also automatically convert pre-esm modules into esm modules!).  Since several of the dependencies in CascadeStudio predate ECMAScript5 modules, this functionality could be a huge time-saver...

UriShX, 2020-11-07:
Cool, I didn't know about Snowpack.
If it's ok with you, I could give this a go.

zalo, 2020-11-09:
Fine with me  👍  

I'd recommend starting from either #15 or [`feat-modules`](https://github.com/zalo/CascadeStudio/compare/feat-modules) (since they have my WIP typescript support integrated).

UriShX, 2020-11-12:
I started working on #15, by separating the service worker `importScripts` from the node_modules and post-processing `opencascade.wasm.js` to exclude `export default opencascade;`. I did that since they are not module imports, and will not be detcted by Snowpack, if I understand correctly.

You can find it in my cloned repo on the [separate-nod_modules](https://github.com/UriShX/CascadeStudio/tree/separate-node_modules) branch.

I included a npm script in `package.json` called 'prep' to clone OpenCascade and opentype.js (which I also added to `package.json`). I cloned TS manually since I believe there's a compilation step, which I did not do.

When I run it locally with the node_modules from the cloned repository it runs OK, but when I try to install all dependencies from npm and then running my pre-processing script `npm run prep`, I get errors regarding missing functions. I could not trace the origin of the problem, but I assume that you've done some more editions to `opencascade.wasm.js` which I caould not easily track. Maybe a different OC build?

I hope I'm on the right track, and hope you can help me trace the bugs. Here are the error logs from Chrome's console:
>CascadeStudioMainWorker.js:21 Null Shape detected in sceneShapes; skipping: {"ptr":7989688,"hash":970850169}
console.error @ CascadeStudioMainWorker.js:21
combineAndRenderShapes @ CascadeStudioMainWorker.js:131
onmessage @ CascadeStudioMainWorker.js:87
CascadeMain.js:339 Generation Complete!
CascadeStudioMainWorker.js:106 Uncaught TypeError: transformation.SetRotation is not a function
    at eval (eval at <anonymous> (CascadeStudioMainWorker.js:44), <anonymous>:478:24)
    at CacheOp (eval at <anonymous> (CascadeStudioMainWorker.js:44), <anonymous>:28:20)
    at Rotate (eval at <anonymous> (CascadeStudioMainWorker.js:44), <anonymous>:475:19)
    at eval (eval at Evaluate (CascadeStudioMainWorker.js:104), <anonymous>:11:17)
    at Object.Evaluate (CascadeStudioMainWorker.js:104)
    at onmessage (CascadeStudioMainWorker.js:87)
eval @ VM8:478
CacheOp @ VM9:28
Rotate @ VM8:475
eval @ VM12:11
Evaluate @ CascadeStudioMainWorker.js:104
onmessage @ CascadeStudioMainWorker.js:87
setTimeout (async)
Evaluate @ CascadeStudioMainWorker.js:106
onmessage @ CascadeStudioMainWorker.js:87
CascadeStudioMainWorker.js:18 Uncaught Null Shape detected in sceneShapes; skipping: {"ptr":7989688,"hash":970850169}
(anonymous) @ CascadeStudioMainWorker.js:18
setTimeout (async)
console.error @ CascadeStudioMainWorker.js:16
combineAndRenderShapes @ CascadeStudioMainWorker.js:131
onmessage @ CascadeStudioMainWorker.js:87
CascadeStudioShapeToMesh.js:140 Uncaught RuntimeError: INTERNAL OPENCASCADE ERROR DURING GENERATE: function signature mismatch
    at IP (<anonymous>:wasm-function[13908]:0xd182a4)
    at invoke_ii (http://127.0.0.1:5500/js/CADWorker/required/opencascade.js/dist/opencascade.wasm.js:11:601670)
    at <anonymous>:wasm-function[18089]:0x11809f9
    at oL (<anonymous>:wasm-function[14167]:0xd1bcfe)
    at invoke_vii (http://127.0.0.1:5500/js/CADWorker/required/opencascade.js/dist/opencascade.wasm.js:11:602653)
    at <anonymous>:wasm-function[18090]:0x1180bd2
    at nL (<anonymous>:wasm-function[14168]:0xd1bd08)
    at invoke_vi (http://127.0.0.1:5500/js/CADWorker/required/opencascade.js/dist/opencascade.wasm.js:11:601514)
    at <anonymous>:wasm-function[18093]:0x1180f31
    at Op (<anonymous>:wasm-function[12655]:0xc3971a)

zalo, 2020-11-15:
If you're trying to do a fresh `npm install`, you may be grabbing the embind version of `opencascade.js` (which is now the default in that repo, but not in CascadeStudio), which adds and renames nearly all of the functions.

I have a mostly working embind port here:
https://github.com/zalo/CascadeStudio/pull/12

You'll notice I also [patch out some lines (and patch in some debug/error reporting functionality).](https://github.com/zalo/CascadeStudio/pull/12/files#diff-7d888ae3a42f72ffc3319a8a7a9476c873382cdfa729f59ef60b2c5ea257d44cR74-R97)