zalo/CascadeStudio #34: CascadeStudio + CadHub 💪

Irev-Dev, 2020-11-16:
Hey @zalo,

First off thanks for putting this project together. As a big fan of OpenScad as well as being a javascript dev, I'm very excited at the prospect of js CADing, and using openCascade under the hood with wasm is very neat. I think filleting, and STEP file support alone are enough for me to buy into this.

I've been thinking about a community hub for code-cad for a while and seeing CascadeStudio made me pull the trigger and start putting something together. I don't like talking about ideas without some kind of proof of concept so here it is.

### [CadHub](https://cadhub.xyz/)

[repo](https://github.com/Irev-Dev/cadhub)

Expect bugs on the above, but you should be able to click through to a part, then "open IDE" to open up CascadeStudio for that part.

Long term I have hopes for CadHub becoming a centralised hub for all things code-cad, as I think the community is a little too dispersed atm (did you know there are this many [code-cad projects!](https://github.com/Irev-Dev/curated-code-cad)). But that's pie is the sky stuff. For the time being I'de like to make the CascadeStudio integration really solid and I think trying to form a community at the same time is a good way of increasing it's impact.

Anyway all of this is just to show you and ask what do you think?
Are you happy for CadHub to use CascadeStudio like this?
Are you interested in collaborating at all?
Any other ideas or thoughts?

Obviously I'm happy to lend a hand, I noticed in #32 the discussion around trying to get the project bundled, I might be able to help there, but three people might be too many.

Irev-Dev, 2020-11-16:
btw one of the next things on my todo list is to make the CadHub homepage a little more "landing page" style, explain it's purpose and whatnot, when I do that I'll make sure that it's not lost that CascadeStudio is seperate project and link it.

zalo, 2020-11-18:
While the [cadhub homepage](https://cadhub.xyz/) is giving me a 502, the screenshots on the repo homepage look awesome!

[image]

It looks like you found the PR where I began the firebase gallery, but I'm not thrilled with the complexity of my integration there (and have kind of abandoned that line of work myself; I'm just not that much of a front-end kind of guy 🙃 ).

The way you're using CascadeStudio looks great; I have no issues.   Feel free to submit a PR or request for any functionality that could make your life/integration easier :-)

Irev-Dev, 2020-11-18:
Hooray 🎉 
Really glad you like it 😅 

That 502 is really annoying, just as I go to show someone. Weird bug, seems intermittent (anyway that neither here no there).

Thanks for that cannibalize link, that's definitely something I can re-use. I'll definitely have to have a good look through that PR.

Having dug through some of the CascadeStudio code, I've got a couple of simmering ideas of things to put PRs up for, though I'll likely double-check with you beforehand.

I'll close this issue now but I'm sure we'll stay in touch. Feel free to reach out if you have any ideas for Cadhub too.

closed 2020-11-18 by Irev-Dev

Irev-Dev, 2020-11-19:
Hey @zalo 
FYI, you should be able to visit the website now.
Turns out netlify made a breaking change and I had to update a few dependencies to get it working again.

Still bugs about but it loads now at least.