zalo/CascadeStudio #37: CascadeStudio parameter sanity checking

udif, 2020-11-17:
It seems that CascadeStudio is not doing any sanity checking on parameters. As a result, a call such as:
```
Sphere(-10)
```

Will result in:
```
Line 75: Uncaught ReferenceError: ___cxa_is_pointer_type is not defined
Line 18: Uncaught There were no scene shapes returned!
```

When this is a part of a larger script, and the parameters are due to long calculations, it is very hard to debug the scripts.
This is an example only, there are other unchecked calls.

brad, 2020-12-10:
@udif As a workaround until something more robust is in place, you can simply log the computed radius (and anything else you want to debug) to the console using `console.log` commands

udif, 2020-12-10:
Yes, I am making use of console.log() and alert().

zalo, 2020-12-18:
Unfortunately, any errors that happen within the OpenCascade kernel are undebuggable due to the way it was compiled. :(

There's [a PR for an Embind recompilation of OpenCascade (which should have better error reporting)](https://github.com/zalo/CascadeStudio/pull/12), but I haven't had a ton of time lately to work out all of the kinks in there, so it's been sitting in PR Queue hell since September...

udif, 2020-12-18:
What I implied by this issue was to add individual sanity checks in OpenCascade for each command before it is issued to the OpenCascade kernel library.
The negative radius for Sphere() was an example.

In any case, the major issue I'm having is #38. If this was solved, there would be less need for issues such as this one (although it is pretty useful by itself).