zalo/CascadeStudio #43: Execution via node.js, is it possible?

brad-jones, 2020-12-10:
This might already be possible but I couldn't see any docs or obvious entrypoint CLI tool to do this? Can WASM be used with node???

Say I have a JS/TS script that defines my model can I use node.js (not the browser) to then export an STL/STEP/SVG/etc file?
Similar to the example given by CadQuery https://cadquery.readthedocs.io/en/latest/installation.html#test-your-installation

I like the idea of using say VsCode on one screen to edit the JS/TS script and then having a simple STL viewer open on another screen watching for changes. This is similar to how I can use both OpenSCAD and CadQuery.

Other use cases would be the ability to have a CI/CD pipeline that say automatically generates various exports like an STL, 2d plans, bill of materials, slices it into gcode if your 3d printing & then creates a new github release with such assets just like you would if it were a traditional software package that your pipeline was compiling & releasing.

zalo, 2020-12-18:
I believe node.js can execute WebAssembly, but I've never used node.js so I wouldn't know where to start 🙃

A barebones cli version of CascadeStudio that just uses the StandardLibrary and the WASM to generate .stl's would be interesting though...

Irev-Dev, 2021-01-06:
I think VS-code extension and generalising this all a bit more, node etc would be awesome.

I'm tempted to attempt so of this. I can't in the immediate future.