zalo/CascadeStudio #44: Interface questions

borioan, 2020-12-10:
First of all I would like to thank you for developing this project!!
I have some questions about the interface and how to display 3D models:

1. I want to know if you can change the type of camera from perspective to orthogonal
2. If you zoom out - respectively the object moves in size, in a short time it disappears from the scene - can this aspect be remedied?
3. It is possible to display a scene world coordinate system?
4. It is possible to change the colors/size of the grid scene and the scene color: for example the background scene color to be white or less darker than actual color - best for me a subtle grey gradient? 
5. It is possible to change the rotational movement of the camera? For example [Helix Toolkit](https://github.com/helix-toolkit) uses turntable/turnball/trackball rotational styles and inspect/walkaround/fixedposition CameraMode.. 
Thank you!!

zalo, 2020-12-18:
1. [Orthographic is possible, but it looks weird with the current shader](https://github.com/zalo/CascadeStudio/blob/master/js/MainPage/CascadeView.js#L27-L29) (and doesn't serve much use since the GUI isn't used to draw things)
2. Is this about the fog?  The fog should move out to cover the size of any object; what is the specific issue?
3. It's possible, but I haven't done it.  [The current grid has one line every 100 units.](https://github.com/zalo/CascadeStudio/blob/master/js/MainPage/CascadeView.js#L66-L71)
4. It's not configurable right now, but [easily changeable in these three lines in the source](https://github.com/zalo/CascadeStudio/blob/master/js/MainPage/CascadeView.js#L23-L25) if you want to pull and run it locally.
5. Sure, I suppose the existing `OrbitControls` could be replaced with [`Trackball Controls`](https://threejs.org/docs/index.html#examples/en/controls/TrackballControls), I don't like it as much, personally.

borioan, 2020-12-18:
Thank you very much for your answers! I like to use just a part of your code, only to display 3D imports files - STEP files, so I have to remove all unnecessary stuff like layout, editor, an so on.. right now I struggle to simplify the interface of my application based on your project. In the future maybe could be useful for other users which need only to display CAD files in the browser to post a derivate and simplified project from CascadeStudio !

zalo, 2020-12-19:
The original example for OpenCascade.js has close to the minimum code necessary to render .STEP files (without outlines around faces):
https://donalffons.github.io/opencascade.js-examples/demos/bottle%20-%20basic/
https://github.com/donalffons/opencascade.js-examples

closed 2020-12-19 by zalo