zalo/CascadeStudio #57: Lofting "twists"

johnsonm, 2021-01-12:
For fun, I have tried out using `Loft` to try to [model a rhodonea (rose) curve](https://zalo.github.io/CascadeStudio/?code=hVZdb9owFH3nV1z1ybQpJKm6D9FO22OlTarUl1YVmgwYYhHszHFG0mn%2FfffaCeSjsEgQ%2B55zru17TwLTKTyoPJNGrGBRwYsulMhz%2BJZyXUi42%2FItN9rqr81gkuidSKUqyokS9gssi9zqnXzji1TgxPwWsBFKGG61CWBRWDBC7rJU7ISyuMZe2gRSvbZSbUajVFjgcA9PqVwJwy74RQC3AUT4HQaw5mkucDaeOeKiRVwg8eMpYqalsnmL7QMoicIw9BIaDZU2kcutK8BRfIih%2FmYSu0WjRlrrVoavbUvj5siPiR8O%2BYmQm6Qt8AHa4edJ1D1XLTF8JYv2vnyANnXrF4jClmS0LtTSSq3AcsXKMfwZAV5G2MIo%2BMFtMnEAXAJzs8cHmEL0KRyj%2BO9RncsTagf8T73U%2BftqB5xQ02nzX4bK41fCcdPYfRPE4cz7R1pyG65Gxbn5EGLWhadLtTTOdwi0WNPaIHXLdfaz3faYdlX359IVzzVzXLdhixSOKRbtEj8z2TsknW%2BLeoxfUg0BGZ3KvJyX4GQgUdrsSlahN6p4DF1xFcE1hgf8ipXIL%2BPeWmWM9DIa0FmJyXtcKj%2FDchOGea6oC6yicW97JTHcJv0tgH1%2FWVQzx8AK0h1Puu9lqSi32%2FqpLFWTpepncR1a86WgRr7OfcfIMuzY4Wn9JI1xFB8tQKw9FoV1%2FDAdcBPk1e5oMs1Ga22AOcchGs7wdtcyHM2v7o9ubI7jBDF5Ew90QGcHrAwRe3YuaEIVhV46oTLyrLhNizytHXNlJ3ltI3xXVFEPrxocbYN42ccb2PcwrPsUtliZX6JkDdrcw8D16MDzS1VuHz7Luzx1Jt91h3gmYYeYvZ1J6ZzQIZ9J2yerc5mvh%2BxzO%2B6zydTkaReg61W55KQJ58ExnFE4G4bfXPyNgKQNKAeoDjB33%2B45mmRFnrBHnVYbrRiF8NymEP7Zn04htxzf1FytQOCH00%2BogJzvhNOPWknc8DWco%2FJpyVPB%2FNMTwHf8N%2BDRQ%2B5%2F&gui=XY8xD4IwEIX%2Fy80NuaLE2MXBWQcGF8NQy0kbYyFQ4mD8714hRcNyuff63evdG0z77FpPPpz1k0CB0YPRNV0cvUDAiQZb0gAKM7moUvuG0StmKEVeCThqY%2BkAKvQjCSh17UYe2WDq00COYovMa1AF12RLUUT3BmrHdeV2rfOB03JETCohBQrJdsSCdebhaYj%2FZvmf%2FsXJyNW9vgdOi8zUL8fM75ZcYxmQ%2B3jwrFYb9em%2BIvULMO1Tfb4%3D).

At many places, always aligned with one axis (Y I think, but without an orientation cube / vectors it's hard to tell), the loft is inverted:

[image]

I don't see a math mistake in my model, though it's possible that one exists. The debugging I've done so far hasn't found one.

I have this modeled well enough for my uses in OpenSCAD, I just thought `Loft` would be a more natural way to describe this shape than lots of polyhedron segments.

zalo, 2021-01-21:
Hooee!  That's a crazy curve!  I suspect the loft function may be having trouble trying to fit a 2000-degree bezier curve to the path...

1) I don't believe Lofts can form closed loops like that
2) [You can't create faces with self-intersecting edges...](https://zalo.github.io/CascadeStudio/?code=hVVRb5swEH7Pr7j1ybSUAFO1TWmnddI29WFSpe6hVRRNhJhghRgGZoFN%2Fe%2B7syFxoOnyEOy77%2FvO9n2Y6RTuZFWIkq9g2cJTXkteVXCbRXkt4HoTbaIyV%2FmnfuCl%2BZZnQtaNJ7n6CHFdqXwr%2FkTLjOOk%2FM1hzSUvI5WXLixrBSUX2yLjWy4V1tgJlUKWJ0rI9WSScQUR3MBDJla8ZGfRmQtXLgT477uQRFnFcebMNHBpAZcIfHcKWORCqspCmwBSAt%2F3DYVGY6ZKRbzRB3Ag72PIf%2BuFumjQUzveqowSZXH0HPEh4f0xPuVindoEE6AVfvCC4311lDJaidpelwnQoq5MgcC3KJOklrESuQQVSdY48HcC%2BCu5qksJ3yOVejoB58D07P4OphC89x0kPx%2FYlTjB1on%2FseO8epmtEyfYtNvqV0nHYyrhuG%2Fsrg%2FicGb8IxS5DavR4aAIqi4NXMi41L7DhIWadgbpWp4XP%2B22h7Sqrj%2Fn%2BvB0M52uDRuERCixtI%2F4kYnBJml%2FG%2BRj%2FJzOEBBxdDJPr1NwMqLIvNw2rEVvtKEDx%2BQ2gEsMj%2FAtaxDfhINaTYjwJhjBWYPiAywdP8PjphzqXFAXWEvjwfIaQuhFmocLu2FZZDONwBOkJ%2B50N1BpSVsv%2FZRK26u0QxXdoSSKOTVyvjAdI8uwQ4en3Zvk4Cg8WIBQOzwUduSH6QibIq5zR69kO%2BEbV3fotHsyGBMuWhDvQheqOMq4Y3ahnRmSJXEfe4vO9rnGx9yjbn4fain0dBRqAoMKbVhgYHZMnzbRO%2FfgFdEGg3zb59EtmG%2BG%2BT5tWud37fEtVGFKNKzP9k%2Ff1a3Z40ypVq%2FDqLyIk6%2FoXR4BXxHcA0XC3uhGdB2wvDSnSuemP65W6yYLQ33meJ2OWYXNKsYs8mJ3BWHZhwI%2FmPy%2B%2FyrNFy7ktXohPpskeQlMEzHgz%2FBxbV1dNL%2B4OZimfzFGVbxaVqlIFBv6UZU1B3x36aqF6RRuVyu0P3zma9SgTzLJjRbnFXWVYmIk130%2Bh3Jf5Kp%2FG43WhquY3hzJd2AmbLTmub%2FAfll477NJ%2F8jHYMfDGox2M%2BA8qAivq9EOSNzSG%2BVP6n3F26SL%2FwM%3D&gui=XY%2B7DsIwDEX%2FxXNUOYUIkYWBGYYOLIjBpIZE0BSRIgbEv%2BOAUhAZLF%2FnXD8e4Pru0keOw5o6BguOkqOWN4HvoGDFyTecwGKlR9VQPAq6xQq1qncKluQ8L8Ae6JxYQUNtuIlngiUvjhrVFMVAYI3EUtbK5Ooe7EziX%2FXShzhIN4NYRCEMKo3yhBp8cKfIKY%2Bt6h%2F97aYz117pMICtM%2FPOx2M%2B%2F57D0Qug5%2Fngj%2Fpb6FrOMyUfgfc%2Bu%2BcL)
3) It might be easier to make a Pipe instead of a loft...

johnsonm, 2021-01-22:
I do weird things, eh?

Of course the approach of using a set of polyhedra as I [used in the OpenSCAD model](https://www.thingiverse.com/thing:4708815) would work here too. (Whether OpenSCAD closes the mesh in that model seems to be a matter of floating point luck; merely changing the radius and leaving the other parameters alone can be the difference between a valid manifold or not.)

Also, I could loft it piecewise. I could make it piecewise by local min and local max distance from origin to control point; then each individual `Loft` would not be self-intersecting, and with the example parameters with five "petals" would result in 10 lofts. That would be easier to do in CascadeStudio than in OpenSCAD, because of the limitations of the declarative OpenSCAD language, even if OpenSCAD provided `Loft()` (which as far as I know it doesn't).

**Would you expect lofts that are exactly butted to each other because their adjacent ends are defined by the same polygons to properly union into a manifold, since they should intersect but not be self-intersecting?**

This could also solve the source of the double-covering problem: Different rhodonea parameters (and also epi/hypo cycloid/trochoids to which this can be reasonably expanded, as in the original from kakaroto) give different periods, and if I used this approach I could check each new face for whether it's within epsilon of the first face and use that to stop the iteration and cover all the reasonable rational versions without explicitly coding the range of the input variable. (Within epsilon to account for floating point inexactly representing pi.)

No matter how this is modeled, handling the self-intersecting nature of the entire curve is the core difficulty. If it weren't for that, it might be easiest to write code to write raw ascii STL! ☺ I could just emit all the outer faces and omit the end faces in STL. But it's not clear to me whether meshlab has a function that could generally and simply resolve those intersections in the "obviously intended" way.

johnsonm, 2021-01-22:
Well, [that was not a rip-roaring success...](https://zalo.github.io/CascadeStudio/?code=jVdbj9o4FH7nVxxV2irMcEmouhcxVLuPlXalStVKHSFUGTDgIdjZXGaSqea%2F7znHceKEgSkPYJ%2FznftnJ0yn8FlniUrlFtYV3JtCyyyDv2JhCgV3R3EUqcnNn24xOZiTjJUuyomW%2BSfYFFluTupZrGOJm%2FRRwl5qmYrcpCNYFzmkUp2SWJ6kzjHGk8oPEJtdrvR%2BMIhlDgIW8DVWW5kG78S7EXwcQYTf4Qh2Is4k78LhnLFrD7tG7G9XsIlROs88AytAq1kYhtYqCnnpjMPWOj%2BozZFb0TpoZOjjw2TG0aLWPJzQvrbfpmKXe7a8p9hkF162O0i1P%2FiGVoCW0R%2BM65TbMU3FVhV%2BvlZAyX60AaPwrFODXaE3uTIacqGDcgg%2FBoCfVOZFquEfkR8mrIAbCHj35TNMIfo9HKLxS2udqQvWrHjLemOy161ZccGaas7%2BS6lZNhKu3eifnBCXc8szlRMrMRq16MOvIXpdW7hMMhVjFgvsJ%2FIhslKlNymzFuWe7bQmVk0Tk3z3qTKjXOsZ3nBLefDDekRHhAh0sfYb%2Fy1QvdKp6iPao%2FyGOguI6PTr%2FroJbs5MtElPZVDh2KvZELrGVQRjFJ%2Fhq6BEfDnrxSpnCC%2BjM3hQovMeloYS4BBIh35uaTZBReteeiUhOEn7M4Knfli0DhiBHaRfrPSp56Ui35z6JS%2BV81Jd9LJVGU5uI7n6ijtgu3ahstmYgG11s3EVNRXy3HdiI4key5XlAdEzaHkzrc%2FuEFezlliEesJWBx2WTc%2BwB8TVnHOerMKkav89KUNiti%2Bp%2BhJ9htEOM9iZFAI%2BECzAnzv%2FPNxgFii7XbQHBhsF0ynggyEB%2BYiCQsRxZZ8G1HHa4QUrYZ1KccRHQhyDSBIU7lSa5dxkDjijo4ddbTzPGx0n%2FI1J7kSc8X1HVEYWNfNhkYX5MmYVmdenJKS59%2FSV0%2BOpQH3Z1zu1pWhY0zD0UHYSxPRa637DEVOwwdlQFedhvbyK01f8jTvAKw4boNpBwDydxFLv8Sm9WOC8Henp49EJv%2Bc9OcfA757cJXkmr3Oq5S8g8bHkBaN0moPoIo%2BaWCPgLa6GyEd3g79%2FD10b3dhorpk71LHxC6QPEhePWgaZtJcBLjfIc3qi4utLbmAtmbuZOEkQGa%2BVVrkSsbPp%2BGOOt7W%2FDNpvnvXzlSHyJeAT4%2FnKIPtgfc3z%2BBx9jSN9NPGErrOmrKXrLL5VrEat2A2pL35m%2BTMpDr5Cs0J3FCsb13IzKbJDsGyGHPLLFNngUEeMWbXXbib3dGu4mxdvMnuRLeprzKc7Sm5vHRmQBAeht%2FhK%2B5SKZIxvvoXetmNI5SNd4HQ1dU7MGN%2FF4Jeu27bDssyd1QVYcwSXarUMV%2FDJYpYUkAXI79f05JoEPpexhNhskJQnUapTcWoUrim2le62fHk1gbs3Erj7qQTwePxMAi8DNNnEJrMHLDYmGXhDt5FCGm%2FXg9%2FHMd3KdF5NLCex2QcOSq%2B60ynNgf58eIyALiUa1%2B1IO8zwaFX7YG46Pw8L5wA7NIeHu3Z7G5Gg9eS1wtbxxcTV3ui60IfVMlrhqU4LOezMCKugCqzNV2ywDOwzfwR%2Fo8JV7Exr27eBcx7AvxovsIAjDOf%2FAw%3D%3D&gui=XY8xD4IwEIX%2Fy80NuaLE2MXBWQcGF8NQy0kbYyFQ4mD8714hRcNyuff63evdG0z77FpPPpz1k0CB0YPRNV0cvUDAiQZb0gAKM7moUvuG0StmKEVeCThqY%2BkAKvQjCSh17UYe2WDq00COYovMa1AF12RLUUT3BmrHdeV2rfOB03JETCohBQrJdsSCdebhaYj%2FZvmf%2FsXJyNW9vgdOi8zUL8fM75ZcYxmQ%2B3jwrFYb9em%2BIvULMO1Tfb4%3D#) It did solve the double-covering problem (confirmed with `console.log`) but avoiding self-intersection for each `Loft` didn't make things better:

[image]

I [also tried to `Union` a list of `Loft` objects](https://zalo.github.io/CascadeStudio/?code=jVdtj9o4EP7OrxhVuirs8pJQ9V7EUt19rHQnVapO6gqhyoABL8HO5WU32Wr%2F%2B82M48QJC1s%2BgD3zPOPxzGMnTKfwWWeJSuUW1hXcm0LLLIO%2FYmEKBXdHcRSpyc2fbjA5mJOMlS7KiZb5J9gUWW5O6lmsY4mT9FHCXmqZitykI1gXOaRSnZJYnqTOcY0nlR8gNrtc6f1gEMscBCzga6y2Mg3eiXcj%2BDiCCL%2FDEexEnEmehcM5Y9cedo3Y365gE6N0nnkEa0DWLAxDy4pCHjpy2LLzg9ocuRRtgMaGMT5MZrxa1NLDCc1r%2FjYVu9zj8pzWJl54mXeQan%2FwidaAzOgPxnW226GmYqsKP19roGQ%2F2gWj8KxSg12hN7kyGnKhg3IIPwaAn1TmRarhH5EfJuyAGwh49uUzTCH6PRwi%2BaVlZ%2BoCmx1vsTcme53Njgts2nP2X0rFsivh2LX%2ByRlxOLc6UzmpElejEn34NcSoawuXSaZizGKB9UQ9RNaq9CZl1aLd405rYdUyMcl3XyozyrXu4Q2XlBs%2FrFt0RIjAEGu%2F8N8C1ds67fqIfLTfUGUBEZ163V%2Bn4OSMok16KoMK217NhtAlVxGM0XyGr4IS8eWst1Y5Q3gZncGDEoP3sNSUAJtAPoxzS70JKhr30isJwUnanxE89ZdFdsAIrCD94k6felEqis2pX4pSuSjVxShblWHnNpJ3X3EFbNUu7Gw2JmC7u9m4ipodct93YiNJHsuV1QHJM2h1M63P7hBHs1ZYhHrCUgcdlU3PsAfE1ZpzkazDpGr%2FPSlDUrZvqfoWfYbRDjPYmRQCPhBswJ87%2FzzcYBZou120BwYLBdMp4IMhAfmIhkLEcWWfBlRxmuEFK2GdSnHER0Icg0gSNO5UmuVcZF5wRkcPq9pEnjc%2BTvgbi9yZOOP7jqmMLGrmwyIL822sKqLXpySkvvf8lfPjqUB%2F2fc7t5VoWMsw9FC2E6T02ut%2BwxFLsMHZpSrOw0Z5FaevxBt3gFcCNkC1g4B1Ooml3uNTerHAfjvR08eTE37Pe3ZeA797dpfkmb3Oqba%2FgMTHkrcYpdMcRLfyqFlrBDzF0RD16G7w9%2B%2Bhy9ENR%2FOeuUIdjr9B%2BqBw8ahlkEl7GeBwgzqnJyq%2BvuQG1pK1m4mTBJHxWGmVKxE7Ticea7zd%2B8ug%2FeZeP19pIl8CvjCerzSyD9bXIo%2FP0dc00keTTug6a7a1dJXFt4rVqDW7JvXNz2x%2FJsfBd2h26I5jZde12kyK7BAsmyaH%2FDJFHGzqiDGr9trN5J5uDXfz4k1mL7JFfY35ckfL7a0TA4rgIPQWX2mfUpGM8c230Nu2Dal8pAucrqbOiRnjuxj80g3bVliWuWNdgDVHcKlWy3AFnyxmSQuyAfX9mp9Ck8HXMm4hNhsU5UmU6lScGocrii2luy1fXk3g7o0E7n4qATweP5PAywApm9hk9oDFxiQDr%2Bl2pZDa243g13FMtzKdVxPLSWz2gYMO67dA%2Buvh6QG6gmgCtw3t6MITVR2DleniPCxcAKzPHB7u2ultRIY2klcIu4svJq72RtfbfFgtoxWe6bSQw06HOH%2FL%2BIrFlYF93o%2Fgb3S43TpizZxO34bSmflX4%2BUV8ArD%2Bf8%3D&gui=XY8xD4IwEIX%2Fy80NuaLE2MXBWQcGF8NQy0kbYyFQ4mD8714hRcNyuff63evdG0z77FpPPpz1k0CB0YPRNV0cvUDAiQZb0gAKM7moUvuG0StmKEVeCThqY%2BkAKvQjCSh17UYe2WDq00COYovMa1AF12RLUUT3BmrHdeV2rfOB03JETCohBQrJdsSCdebhaYj%2FZvmf%2FsXJyNW9vgdOi8zUL8fM75ZcYxmQ%2B3jwrFYb9em%2BIvULMO1Tfb4%3D) but that gives me `Union() encountered ___cxa_is_pointer_type is not defined` so I'm still doing something wrong.

zalo, 2021-01-22:
[Hrm, I just figured _a_ way out.   Try this.](https://zalo.github.io/CascadeStudio/?code=hVVRb9owEH7nV9z65NAUkkzVNlGmtRPrkLoOjW5qVVVTAAMWwc4cZySb%2Bt93thNIAnR5iO2777s7353tbheGPImZpDOY5PAgUk6TBC6jUKQMLlbhKpRCiQ%2FlpLMUaxoxnmYdTtV7mKaJEmv2J5xEFBfyN4UF5VSGSkgXJqkCSdk6juiacoU%2BNkwtIRJzxfii1YqoghDKrw%2FjiM2oJCfhiQvnLvj491yYh1FCceX0DGFygDBBwptjhFgwrpI6wcqQ5Xte8Qs8b5%2Brlmy6MinZcbcypL%2FuBMatX1IL3kyGc9UI0siQE2iOt89ZUrZYqjrHynSc7zp%2BfYMFS4YzljZ2Z2U6vHPrxvcqrNY85VPFBAcVcpI58Lel6ZKqVHL4EqplxyigDcSsRkPogv%2FWc5D8vGMn7AjbKP7HnorkMNsojrD1hpNfUuFerSecl0XelEKc9mxvMaU7Eb3p8qERtDqxcMan0vQkKiqobtEsRfFF%2FLPaAIGOqqhS2yTPlNQpKrFCSIgmJtUU3xPW2KTe3wr5KG%2FrHAIiapl5eJmCiz0KF3KdkRzbIw8cqJNzH85QvIfPSYb4LGj4ygKEZ%2F4enGRovIHV6SeYbq1DO6e6CiTX80Z4mUaYIO3gwqbpFtnEIDCDesSdbhpWcm3bhH7MSl5ayZtWTIV0i5BdRbvF4XFwFhRnEBFFhUtttZrXVA2xW0a6SQhzsY3wrnMhmYYRdWwkprsC3VYYy7bNeltd5qHu3hSwFOVa9FATZb5FBVWYb2FVmcmYphcdgMc89xv6vNRjxVGfNfWl2qbfK1LsVVCxdZGRUluOnmvSu8VZV7mJw1o5iOMv2DurAV8wWAPitRcjUt885pB4HU9%2Ffg%2B6Xbj7PBzDYDQe3ny9hdG3wY%2FB7d0YpQO4%2FX5zA58uPw6MITYnr0xFi1JWGutRh9y2hXZNWNuFdv1kI3mmeMPuc%2BMqNz7GLZs0ifF9pSP7aPXhEfVzIYGYxkKB18PhonJr6fVpf9dr5ZmoGurEabIkzfZVMqWgXz48rzpRl7MZXnpwRReYBv1CFyHFLKbXKb4t6P5qbMySqnVrySlu3akUSTKm9sT0YSSifCE4edzm5dG8tvgGPrl1WXBYtsSEBXsKb6sw8iftfoSBkqp%2Fdxc76v8B&gui=XY%2BxDsIwDET%2FxXNUOYEKkYWBGYYOLKhDSA2JEGnVpmJA%2FDtOUVrEYvkuzxf7BbZ9dG2gEI%2FmQaDBmsGahk6eniDgQIOraACNBcpZVibcmD1PpqoF7I11tAMd%2B5EEVKbxI8%2BsMPd5QKFYI%2FMGdMk121KUyb2A3nD9c7vWh8hpEjGLmWBLISYqOm%2FvgYb0baF%2B9JImE9f05hpBq8RM%2FXzL992RvzkG5LaQWf0t1Ofzytwv%2B3AI1u8P#)

By adding a very small epsilon to the Pipe guide, I was able to sort of bypass the self-intersecting face issue... While the Part itself has overlapping faces, I've verified that the Prusa Slicer can automatically resolve these non-manifold ambiguities and print the part without issue.

(Make sure to change the mesh resolution if you'd like a smaller model.)

johnsonm, 2021-01-23:
A very short basket-weave! Very sneaky!

As it happens, I'm using Kiri:moto (another all-in-browser tool) to render gcode to cut with my router, and it can't handle non-manifold geometry. He doesn't use your approach of compiling C into js to use in the browser; he's sticking to pure native javascript, which as a discipline doesn't let him use the same libraries prusa-slicer is using to resolve the non-manifold ambiguities.

Again, this exercise is just for fun; I got a manifold out of OpenSCAD with adjacent polyhedra and generated gcode just fine, so I'm just trying out new ideas here. If I can get a consistent manifold it might turn into another nice example.

I wonder if using my piece-wise approach with pipes instead of a loft would help? In any case I should drastically reduce my iterations since I'm not making polyhedra. That default just carried over from making polyhedra smooth enough. Changing that was why I filed #61 ...

johnsonm, 2021-01-23:
A [`Pipe` by parts](https://zalo.github.io/CascadeStudio/?code=lVdZb9tGEH7XrxgEaEDZOkgF7gFZQdu3PBQwYBSoIQjBilpJa1G7LA%2BbdOD%2F3plZHkvqSKoHiZyZb85vh9R0Cl90GqtEbmBdwpPJtUxT%2BCMSJldwfxAHkZjM%2FF5fTPbmKCOl82KiZfYZwjzNzFG9iXUk8SZ5kbCTWiYiM8kI1nkGiVTHOJJHqTOM8aqyPURmmym9GwwimYGABTxGaiMT74P4MIK7EQT47Y9gK6JU8p0%2FnLPt2rFdo%2B0vV2xjo3SWOgArQNTM9y3ozneQfgvN9io8cB9adCNDB58mMw4VtHB%2FQvcVfpOIbeZg%2BZ4CE86%2FjNtLtdu7QCtAZPAb23Vq7UATsVG5m68VULJ3NmDgn7RpsM11mCmjIRPaK4bwbQD4SWSWJxr%2BEtl%2Bwgq4AY%2FvHr7AFIJf%2FSGC31t0qi6gWfE9dGjS82hWXEBTzem%2FCTXLRsLreu6vtRAv55ZkKiNKYjRq0aefffS6tuYyTlWEWSywn77vB1aqdJgwZVHuYKcVqyqamPirS5UZ5VrN8IZbyoMfViM6oIlAF2u38f94qlc6VX1APMpvqLOAFp1%2BPV2H4M0JRJvkWHgljr2cDaELLgMYo%2FjEvvQKtC9mvVjFDM2L4MTcK9B5z5aG4uEQSId%2Bbmk2XknXvfQKsuAk7c8IXvthEe2xBXaQfrHS156Xknxz6pe8lLWX8qKXjUpxcqHk6kvugO3ahcpmYzJsq5uNy6CpkOe%2By%2FFE4uyXK8sDomdLmynMWjaxqsOqRm0StfsaFz4R1ZWUfYk%2BsdG1zWBrEvCY3yzAn3uX3jcYDGW3i5b%2FWDdMp4BLPgb5goJcRFFpNzs1kO6wGgnrRIoDrvcoAhHHKNyqJM24ZxSQk2Ky31S7at7oOL2n8zo1oxOIzW0yanVFYH3OzjoNrNOzSiYSRa0Ohk%2BjHnb1Za3Hg4D6oq%2Bv1ZaVfsU837Gy0yJyV9r61x8x6xo7G6rkPKyXs3b6ir9xx%2FCKw8ZQbcFjak4iqXf4VF4skBM1z%2BnjUA6%2F5z05x8DvnrxO8kRe5VTJ30Hik8gJRuk0Z6%2BOPGpijYBv8WqInK2X9seP0MXoBqO5Zu5QB%2BMWSB8kN562FFJpzz9ehngW6CGKryuZgbVkfqfiKEGkfK20ypSIakzHH5%2BDtvb3QfvNs367MkReAS4x3q4Msm%2Bsr3ken1pf44hrXTdoK0JJ3RGh3QGQp9ggo%2FEKZ8cm1Lu9SaWmRgm7AbBJO36Q8pOzic%2FecCc2jVrWs8JXk9WoFddj74vfWP5GCvvIdbWatfpUu7I1Wd7Hebr3lg2BfH43IyASZgRLe8VxOd1Vu9KrmtJqq%2BNatVt1Ue1U91yh5Pa2Zh02cy%2F0Bt%2BVXxMRj%2FGVOtebdt6JfEGXHu27ztEc43se%2FNR1245SFlmNumDWnPWlWi39FXy2NksKyAI8SOf05JoE7qHBEiITIvuPolDH%2FNgo6qbYvqqKPu9nE7j%2FTgL3P5QAnsMfSeB9gJAwQmLy6Y2MiQcOA2wkn8bb9eD2cUzrnxaDiVBkdl5tSq%2FR6J%2FHp2LpUAK6nGh8tzPtUMPhVeWDpI2f50XtAFs0h%2Bd75%2F42IEnrymmGW%2BLzahmsOmMhx2Fi0vSx2n4LeDBRuTO6gjgxV8vZasioB6zTc2Ej%2BPMxxj%2BFsu5K9Udj2KwQbo3N5X%2Bg%2BcBNp39rtPLYxXD%2BHw%3D%3D&gui=XY8xD4IwEIX%2Fy80NuaLE2MXBWQcGF8NQy0kbQyFQ4mD8714lRcNyuXf97l3fC0zX9p0nH866JVBg9Gh0TRdHTxBwotGWNILCTC6q1L5h9IoZSpFXAo7aWDqACsNEAkpdu4lXNpj6tJCj2CLzGlTBNY2lKOL0BmrHdTXtO%2BcDu%2BWISSSiQEYiE6wzD09jPJrlf%2FrnJSNXD%2Foe2Coy335JMr9bco1lQO5j2lmtvjOkcEXqFyCaYPX%2BAA%3D%3D) works, renders almost instantly, and produces a manifold object according to meshlab. The code is ugly but there was no point in making it better without first seeing how this approach would work.

When I try to `Union` the parts (see the comments at the end) I get `Line 18: Uncaught Encountered Null Face!` I was hoping that a `Union` would refine the shape and get rid of the excess faces, but no such luck.

johnsonm, 2021-01-23:
[I improved (I think) the code](https://zalo.github.io/CascadeStudio/?code=hVdLb%2BJIEL7zK0pzmDUJD5tR9iHCaHcPK81hpEi5bITQyEADDabb68fEzk7%2B%2B1ZVu%2B22MSxSwF1VX72r2plO4YtKY5mILaxLeNG5EmkKf0ShziU8nsJTmOhM%2F24fJgd9FpFUeTFRIvsMmzzN9Fm%2BhetI4CH5LmAvlEjCTCcjWOcZJEKe40ichcrQxqvMDhDpXSbVfjCIRAYhLOA5kluReB%2FCDyN4GEGA3%2F4IdmGUCj75wznLrh3ZNcr%2BckM21lJlqQMwBETNfN%2BAHnwH6TfQ7CA3J85Dg65pqODTZMamggbuT%2Bhc4bdJuMscLJ%2FJMOH867iDkPuDCzQERAa%2FsVwr1hY0Cbcyd%2F01BHL2wRgM%2FIs0DXa52mRSK8hC5RVD%2BHcA%2BElElicKvobZYcIMuAOPT09fYArBr%2F4Qwe8NOpVX0Mz4P%2FRGp%2F1oZlxBU8zpPwkly1jCZ1v3V0vEx7lpMplRS6I1StGnn33UujbiIk5lhF4sMJ%2B%2B7weGKtUm4ZZFuoOdVl1VtYmOv7mtMiNfqxrecUq58MOqRCcUCVHF2k38357shE5RnxCP9DvKLKBEK18vtyF4uIAonZwLr8Syl7MhtMFlAGMkX8iXXoHyxaxjq5iheBFciHsFKu%2FIUlE8LALxUM891cYr6bnjXkES7KT5GcFr1yyiPZbADNIvRvra0VKSbnb9mpbSaimvatnKFCu3ERx9yRkwWbsS2WxMgk10s3EZ1BFy3fc5TiTWfrkyfZCKPXUWdczSr2jUsk0rTWHWdBizWp1Ws3Ui99%2FiwqfmdSlll6IuZJSVGex0Ah73PBFGcGQ6nh7dzr9Dm0i7XzSjgaL395gXmE4BL4EYxHek5mEUlWbzU4LphJEJWCciPOH6jyII4xiJO5mkGeeUjLODPAx31S6b1zx29aWfR%2BUidl02nzdsgX8lrTgrJ2c0yVik2v2GVwTG9qzXeGCM9zLJatBjnVvHsc6NS25Wg%2Bj38EvLx8FjHR2%2BZZsp8KtOd2M0nUDDVHHtrz%2FiLq%2FljKmS%2FTBaeuXUDX3jluANhS3BcLv9K9zQMPBF1DB2hkojQiS5A4%2FHZhIJtcc3hsUCm9LOIH2c1sfveYfO%2FuB3h24DuqBX%2Fjv0xtMsyStH30Gg044TG5wMHaGTeu8dhw2Y3PfqnrCujmrnRsBHfBrilFU30BA%2BfoQOStUoxSnlArRRblLog8OImyLFLWP2GT6Sm%2FRSgK9fmYa14HlMw7OAMOVnqWQmw8hi5rDV6ieuFpyxjKJlgAe5ifX9MmofPpultzyOgxXuOIrMMnhkhvDjB1Syj9dkHyvZnggjvUF3z%2Bj3OT9jguAcFvTYkrN7dhLn6aFVn2sFbuJ5r9uwknOdQAe2YieVgCcdlXtMMu3QJxk3ieIxe7sxP7zZHZcM4MYc9QHULQvjfsStUe0i7FS2Ere0XYirbjVqs2xb97HemPdGTPOO1JVQLKH6JarFYOpie5z9w99wY64dyFPsca3wCUvHItT%2BB50KRb0emkvHdoZ5mWONZtlwoyy5%2B0awNHcIh8KWVuZKR%2BObCDXy5ERaxwMHbFrZJ9F2%2B7nbbEyr3d0dVnTYuo4X1S1cK6p24RgCZPDVW%2B9PG1GzQms9x4VVsJSrORwfnfN9QJRGlTM2bjzH1TJYDd0KkOJNotP0uVozCzsLFcSxuVrOVkNG0Yh4LmwEfz7H%2BN%2BksCmo%2FkPht6f%2FAA%3D%3D&gui=XY8xD4IwEIX%2Fy80NuaLE2MXBWQcGF8NQy0kbQyFQ4mD8714lRcNyuXf97l3fC0zX9p0nH866JVBg9Gh0TRdHTxBwotGWNILCTC6q1L5h9IoZSpFXAo7aWDqACsNEAkpdu4lXNpj6tJCj2CLzGlTBNY2lKOL0BmrHdTXtO%2BcDu%2BWISSSiQEYiE6wzD09jPJrlf%2FrnJSNXD%2Foe2Coy335JMr9bco1lQO5j2lmtvjOkcEXqFyCaYPX%2BAA%3D%3D) and it creates the same model with what I think is slightly more readable code.

closed 2021-01-24 by johnsonm