zalo/CascadeStudio #61: Slider with realTime false evaluates instantly, hard to change values in UI

johnsonm, 2021-01-22:
It's hard to use the sliders.
* Setting `realTime` to false doesn't prevent them from firing, making the "evaluate" button useless
* The hover docs suggest a `callback` function not present in the signature.
* Typing an exact value doesn't work; every keystroke results in lost focus and re-evaluating the model with that one character change.
* Using the mouse doesn't work either because it starts evaluating even before I let up the mouse button, so I can't even move the slider where I want it with the mouse.
* I can't figure out how to revert to the default values in code in an existing model, since it appears that the current slider values are encoded in the URL.

Am I missing something?

zalo, 2021-01-29:
This is good commentary; the real-time slider only differentiates between "Evaluate as you drag" and "Evaluate when you let go".

johnsonm, 2021-01-30:
Could you maybe also look into "realTime===false" meaning "typing doesn't count until you hit enter or lose focus" at the same time?

zalo, 2021-01-30:
`realTime` now means that slider changes will only apply after you've finished dragging the slider.

By default, Sliders will now not cause a rebuild (via dragging or typing).  Unfortunately, they will still attempt to enforce their range while typing, which is fairly annoying (and often requires highlighting the text to replace it).  This is a deeper issue within ControlKit, and harder to fix, unfortunately :-(

closed 2021-01-30 by zalo

johnsonm, 2021-01-30:
Thank you so much for the changes! It is so much easier now!