zalo/CascadeStudio #73: Text3D fails when the text contains an "f"

NothanUmber, 2021-02-15:
Played around with the CascadeStudio demo on https://cadhub.xyz/draft
When replacing the text "Start Hacking" in line 17 with any word that contains and f then rendering fails. Works for capital F and other letters I tried.

Irev-Dev, 2021-02-15:
Hi @NothanUmber, I just double checked that I can replicate this issue in CascadeStudio proper, to make sure It's not a regression I introduced on Cadhub, and its a bug common to both.

I also figured that the bug seems to be specific to the default font `Consolas` which means if you need a work-around you can use one of the other fonts.
```
Translate([-25, 0, 140], Text3D("abcdefghijklmnopqrstuvwxyz", 36, 1, 'Roboto'));
Translate([-25, 0, 100], Text3D("ABCDEFGHIJKLMNOPQRSTUVQXYZ", 36, 1, 'Roboto'));
Translate([-25, 0, 200], Text3D("abcde ghijklmnopqrstuvwxyz", 36, 1, 'Consolas'));
Translate([-25, 0, 250], Text3D("ABCDEFGHIJKLMNOPQRSTUVQXYZ", 36, 1, 'Consolas'));
Translate([-25, 0, 300], Text3D("abcdefghijklmnopqrstuvwxyz", 36, 1, 'Papyrus'));
Translate([-25, 0, 350], Text3D("ABCDEFGHIJKLMNOPQ STUVQXYZ", 36, 1, 'Papyrus'));
```

I actually found a similar issue with `R` for papyrus.

I wonder @zalo if it might be worth changing the default font to roboto since it seems to be the most reliable?

[image]

zalo, 2021-02-15:
Good idea; the issue comes when the kerning of the font specifies a pair intersecting characters (OCC doesn't like self-intersecting faces).

zalo, 2021-02-15:
Hrm, interestingly enough, catching the error in the face construction process _appears_ to create text containing the problematic character in the problematic fonts.   I'm going to leave the error/warning message there (since there may be second-order bad behavior down the line arising from this beyond what I've tested).

"Fixed" in d20bfd329f5caa6ab656b351aa6f8741c1b0f541 ; [visible on the main CascadeStudio deployment](https://zalo.github.io/CascadeStudio/?code=CylKzCvOSSxJ1YjWNTLVUTDQUTAxiNVRCEmtKDF20VBKy0xJyalU0lEwNgNK6hkClag75%2BcV5%2BckFqtraloDAA%3D%3D&gui=q1ZKzs8tyM9LzSvxS8xNVbJSSk4sTk5MSQ3LTC1X0lHyTS3OCEotVrIy0DOE84IS89KBSqMN9AwMdYxidZScE5MzUu2VrEqKSlN1lIISUzJLgVqMDWBsmAYjAx0Tg9haAA%3D%3D).

closed 2021-02-15 by zalo

NothanUmber, 2021-02-15:
Thanks @zalo and @Irev-Dev  for verifying and fixing!
Would this be worth reporting to OpenCascade?

zalo, 2021-02-15:
Not really; the ability to generate Text is functionality I hacked together by manually assembling faces from parsed bezier curves of fonts and text via `opentype.js`.   There are probably ways to repair invalid faces and bodies in OCC, but I haven't taken the time to expose them yet.

NothanUmber, 2021-02-15:
Okok :) In any case hats of for what you have already achieved with CascadeStudio!