zalo/CascadeStudio #100: Questions about performances

antonymarion, 2021-05-30:
Hi all,

First of all, thank you for porting opencascade to wasm, it rocks!

I am having one issue using one of the examples from cadhub.xyz

The example I am using is the following: https://cadhub.xyz/u/chnorzibaert/through-hole-resistors-box/ide

I am testing how the 3D Viewer behave once we increase the value of numberOfCompartmentsHorizontal from 8 to 64 for instance.

And once I use 64 in cascadeStudio (instead of 8) then the loader is stuck at state "UNION"

[image]

Am I missing something or is there any performance limitation to keep in mind?

Regards,
Antony

zalo, 2021-06-09:
OpenCascade wasn't particularly performant to begin with, but the translation to Javascript/WASM has slowed things down a bit.

The linked Cadhub example works on my machine, but I can tell that it's very resource intensive.  It's unsurprising that increasing the workload 8x grinds it to a halt.

It's possible that some newer compilation of [`opencascade.js`](https://github.com/donalffons/opencascade.js/) improves performance/memory stability (due to newerish versions of Emscripten), but it's a small chance.   You can see a bit of my experiences with that [here](https://github.com/zalo/CascadeStudio/pull/12) (as well as find [a beta version that uses this backend](https://cascade-studio-git-feat-embind.zalo.vercel.app/)).

For really heavy projects, you may be better off with a different system (like CadQuery, which uses a C++ compilation of OpenCascade in its backend).

closed 2021-06-09 by zalo

antonymarion, 2021-06-09:
Thanks for your swift reply !

Indeed the beta did the trick on a basic laptop.

cheers