zalo/CascadeStudio #148: Cylinder of same radius having different radius during visualization

lokitkhemka, 2022-07-15:
When, I was running the following code: 

```
function DrawCylinder(Center, Radius, Length, Angle,  Direction)
{
    let shape = Cylinder(Radius, Length, false);
    shape = Rotate(Direction, Angle,shape, false);
    shape = Translate(Center, shape, false);
    return shape
}
let scale = 2.0;
let Radius = Slider("Radius", 0.65 , 0.5 , 1.0);

let RefPipe = DrawCylinder([0,0,0], Radius *scale, 4*scale,0, [1,0,0]);
let Pipe1 = DrawCylinder([8*scale,75*scale,8*scale], Radius *scale, 4*scale, -90, [1,0,0]);
let Pipe2 = DrawCylinder([8*scale,79*scale,8*scale], Radius *scale, 12*scale, 0, [1,0,0]);
let Pipe3 = DrawCylinder([8*scale,79*scale,20*scale], Radius *scale, 37*scale, 90, [1,0,0]);
```

Pipe2 has a vastly different radius than rest of the cylinders. This is on google chrome.

raydeleu, 2023-09-19:
There are strange things happening. I used the following steps: 
* started CascadeStudio as a web application
* copied the code above into the editor window

I then got the following result.
[image]

I then removed the checkmark at the ```cache```, immediately it changes to the following result: 

[image]

Setting the cache again to enabled changes the result again: 

[image]

Another way to see the change is to keep the cache checkmark but to put comment marks around some of the shapes and then remove these comment marks again. I suppose that this also flushes the cache for these shapes. 
My assumption is that your function makes it difficult for the caching mechanism to recognize when a shape has to be recalculated. Leaving the cache off seems to solve all issues.

lokitkhemka, 2023-09-25:
Yes, it solved the issue. Thank you.

closed 2023-09-25 by lokitkhemka