zalo/CascadeStudio #160: Uncaught SyntaxError: Unexpected token 'export'

Flyingbeanss, 2023-10-30:
I’m trying to run it in the vs live server.
After I run `npm install` and install rawflate.There is still something wrong.
[image]
[image]

Can anyone help me? Thanks.

zalo, 2023-11-10:
Don’t run npm install; this repo is from before I learned how the web ecosystem works, so all the dependencies are hard-committed to it.

Flyingbeanss, 2023-11-10:
Thanks for your help. I'll try it.

wuzejinCN, 2023-12-31:
Not using NPM, but it still cannot function properly. What's wrong with it? Help!

zalo, 2023-12-31:
Is it the same error as above?  Don’t use any package manager; just delete the repo and re-pull it without running any package manager…

wuzejinCN, 2024-01-01:
It's OK now. Thank you! This is a good project, come on!