zalo/CascadeStudio #161: [Firefox] Save/Load project buttons error: window.showOpenFilePicker is not a function

oiale, 2024-01-19:
Steps to reproduce:
1. Go to https://zalo.github.io/CascadeStudio/ using Firefox v121+
2. Click 'Save Project' or 'Open Project'
3. Nothing happens, and console shows the following error:
```
Uncaught (in promise) TypeError: window.showOpenFilePicker is not a function
    getNewFileHandle https://zalo.github.io/CascadeStudio/js/MainPage/CascadeMain.js:535
    loadProject https://zalo.github.io/CascadeStudio/js/MainPage/CascadeMain.js:589
    onmouseup https://zalo.github.io/CascadeStudio/:1
```
Interestingly, the button 'Import STEP/IGES/STL' works correctly and shows a dialog.
Turning ublock on/off shows no difference.
Tested on
```
MacOS 14.2.1
Firefox 121.0
```
```
Windows 10 22H2
Firefox 121.0.1
```

hadess, 2024-02-08:
Ran into the same problem trying to load https://www.printables.com/model/425966-parametric-sliding-lid-box
Same symptoms. Worked in Google Chrome in a pinch.

jonasfa, 2025-09-06:
PSA: while it's disabled by default, [Brave](https://brave.com/) supports this browser feature if you enable the "File System Access API" feature flag in brave://flags/#file-system-access-api.