zalo/CascadeStudio #167: Assistance Needed: Translating openScad Code to CascadeStudio Syntax

yumilious, 2025-01-13:
Hello,

I am looking for help documentation or guidance on how to write equivalent code in CascadeStudio for the following openScad script. Specifically, I want to achieve the same functionality in CascadeStudio as the openScad code provided below:

```openscad
outer_length = 74.6;       
outer_width = 38;        
thickness = 17;          

corner_radius = 10;     
corner_radius2 = 19;

inner_length = 56.603;      
inner_width = 20;        

position = [0, 0, 0];    

magnetic_ring(outer_length, outer_width, thickness, inner_length, inner_width, corner_radius, position);

module magnetic_ring(outer_length, outer_width, thickness, inner_length, inner_width, corner_radius, position) {
    translate(position) {
        difference() {
            rounded_rectangle(outer_length, outer_width, thickness, corner_radius2);
            translate([0, 0, -1]) 
            rounded_rectangle(inner_length, inner_width, thickness + 2, corner_radius);
        }
    }
}

module rounded_rectangle(length, width, height, radius) {
    hull() {
        for (x = [-length / 2 + radius, length / 2 - radius])
            for (y = [-width / 2 + radius, width / 2 - radius])
                translate([x, y, 0])
        cylinder(h = height, r = radius, $fn = 50);
    }
}
```

Could you please provide guidance or examples on how to implement this in CascadeStudio? Any help or documentation would be greatly appreciated.

Thank you!

raydeleu, 2025-01-13:
Have you found my manual on CascadeStudio? If you Google this you will find https://github.com/raydeleu/CascadeStudioManual.

Termina94, 2025-03-25:
> Hello,
> 
> I am looking for help documentation or guidance on how to write equivalent code in CascadeStudio for the following openScad script. Specifically, I want to achieve the same functionality in CascadeStudio as the openScad code provided below:
> 
> outer_length = 74.6;       
> outer_width = 38;        
> thickness = 17;          
> 
> corner_radius = 10;     
> corner_radius2 = 19;
> 
> inner_length = 56.603;      
> inner_width = 20;        
> 
> position = [0, 0, 0];    
> 
> magnetic_ring(outer_length, outer_width, thickness, inner_length, inner_width, corner_radius, position);
> 
> module magnetic_ring(outer_length, outer_width, thickness, inner_length, inner_width, corner_radius, position) {
>     translate(position) {
>         difference() {
>             rounded_rectangle(outer_length, outer_width, thickness, corner_radius2);
>             translate([0, 0, -1]) 
>             rounded_rectangle(inner_length, inner_width, thickness + 2, corner_radius);
>         }
>     }
> }
> 
> module rounded_rectangle(length, width, height, radius) {
>     hull() {
>         for (x = [-length / 2 + radius, length / 2 - radius])
>             for (y = [-width / 2 + radius, width / 2 - radius])
>                 translate([x, y, 0])
>         cylinder(h = height, r = radius, $fn = 50);
>     }
> }
> Could you please provide guidance or examples on how to implement this in CascadeStudio? Any help or documentation would be greatly appreciated.
> 
> Thank you!

This is pretty much what you're after

```
let outer_length = Slider('Outer length', 74.6, 70, 400, true);
let outer_width = Slider('Outer width', 38, 30, 400, true);
let thickness = Slider('Thickness', 17, 1, 100, true);

let corner_radius = Slider('Corner radius 1', 10, 0, 50, true);
let corner_radius2 = Slider('Corner radius 2', 19, 0, 50, true);

let inner_length = Slider('Inner length', 56.603, 30, 400, true);
let inner_width = Slider('Inner width', 20, 10, 400, true);

const rounded_rectangle = (length, width, height, radius) => {
    let box = Box(length, width, height, true);

    // 0.00001 bit of a hack to prevent it from erroring with over-sized fillet
    // but you could replace this function with a similar function that creates
    // cylinders for corners with a box in the middle all wrapped in a union
    return FilletEdges(box, radius-0.00001, [0,2,4,6]);
};

const magnetic_ring = () => {
    return Difference(
        rounded_rectangle(outer_length, outer_width, thickness, corner_radius2),
        [
            rounded_rectangle(inner_length, inner_width, thickness, corner_radius)
        ]
    )
}

magnetic_ring();
```

closed 2025-03-27 by yumilious